grbl/grbl #1338: X Axis Travel

Gitfodder, 2018-01-09:
Hi,
This is probably very obvious to you all but as a noob I am a bit stuck. 
I am using GRBL 9.0 on a woodpecker board, if the tool has to travel more than 130mm between operations in the x axis the tool head just stops as if it has run into something and the program goes to "idle" but the spindle is still running. if the gaps between operations are less then everything works as it should.
Any ideas?

chamnit, 2018-01-09:
That is unusual. Is it a hard stop? Typically this type of problem is the result of some kind of hard reset. Like when the USB connection gets disconnected and reconnects, which causes the Arduino boot loader to auto-reset the micro controller. If it's a decelerated stop without any alarm or errors, then it's likely Grbl isn't receiving more g-code for some reason.

Gitfodder, 2018-01-10:
Hi Chamnit,
It is a very definite stop, like it has run into a wall! but if I then click on abort and then send, it carries on as if nothing has happened until it tries to travel back up the table to drill the next hole when it gets to approximately 130mm of travel and then it stops and the connection goes to "idle". I have had a problem with the usb connection when milling a large pocket, after a while the cnc machine loses contact with the computer and I have to restart the computer as the com port seems to be missing until this is done.

X3msnake, 2018-01-10:
What Serial chip does your arduino have? CH340 CP210?

Gitfodder, 2018-01-10:
it's a CH340

langwadt, 2018-01-10:
max steprate?

Gitfodder, 2018-01-10:
Hi Langwadt,
$100=800.00mm/min (x, step/mm)

X3msnake, 2018-01-10:
do you have a standalone CP210 or FTDI serial converter that you can use
for testing. CH are known to have problems with fast serial interface

https://github.com/gnea/grbl/wiki/Known-Issues
https://github.com/gnea/grbl/issues/294

Gitfodder, 2018-01-10:
Hi X3msnake,
I'm afraid not, but having read about the woodpecker board and the problems with the CH340, I have ordered another cnc shield and a Arduino UNO to replace the woodpecker.

X3msnake, 2018-01-10:
while you are at it buy a CP2012

https://www.ebay.com/sch/i.html?_from=R40&_sacat=0&_nkw=serial+converter+2102&_sop=15

chamnit, 2018-01-10:
800 steps/mm is quite high. Please post your complete $$ settings. I think you may be going over the 30kHz step rate limit.

Gitfodder, 2018-01-10:
$$ < $0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=5 (dir port invert mask:00000101)
$4=0 (step enable invert, bool)
$5=0 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=3 (status report mask:00000011)
$11=0.010 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=0 (hard limits, bool)
$22=0 (homing cycle, bool)
$23=0 (homing dir invert mask:00000000)
$24=25.000 (homing feed, mm/min)
$25=500.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=800.000 (x, step/mm)
$101=800.000 (y, step/mm)
$102=800.000 (z, step/mm)
$110=5000.000 (x max rate, mm/min)
$111=5000.000 (y max rate, mm/min)
$112=800.000 (z max rate, mm/min)
$120=10.000 (x accel, mm/sec^2)
$121=10.000 (y accel, mm/sec^2)
$122=10.000 (z accel, mm/sec^2)
$130=300.000 (x max travel, mm)
$131=200.000 (y max travel, mm)

langwadt, 2018-01-10:
5000*800/60 ~= 66666kHz far above the 30kHz maximum

Gitfodder, 2018-01-10:
These are the settings "straight out of the box" which are the settings that need changing? I would guess $110 to $112 but I am new to all this so I am probably wrong! what should the settings be?

chamnit, 2018-01-10:
Try reducing your micro stepping on your stepper drivers. Often its just a few jumpers. This will reduce the step/mm. If you can reduce it by a factor of 4, it should bring it down to 16 kHz. For example, you're set at 16 micro steps. Try 4 micro steps.

closed 2018-01-10 by chamnit

Gitfodder, 2018-01-11:
Thank you for all your help everyone, I have adjusted the $110 and $111 to 2250 and everything seems to be fine now. Unfortunately the usb keeps losing connection which I can only get back by switching everything off and restarting so I will definitely be changing the board now.
One again, thanks for all the help with this.

109JB, 2018-01-11:
I would start by changing the sub cable. You want a thick cable with ferrite choke on it to help reduce noise.