grbl/grbl #1339: error during compilation

mlo300, 2018-01-11:
In file included from C:\Users\Admin\Documents\Arduino\libraries\grbl/grbl.h:43:0,

                 from C:\Users\Admin\Desktop\grbl-master\grbl\examples\grblUpload\grblUpload.ino:27:

C:\Users\Admin\Documents\Arduino\libraries\grbl/nuts_bolts.h:25:0: warning: "false" redefined

 #define false 0

 ^

In file included from C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:24:0,

                 from sketch\grblUpload.ino.cpp:1:

c:\program files (x86)\arduino\hardware\tools\avr\lib\gcc\avr\4.9.2\include\stdbool.h:42:0: note: this is the location of the previous definition

 #define false false

 ^

In file included from C:\Users\Admin\Documents\Arduino\libraries\grbl/grbl.h:43:0,

                 from C:\Users\Admin\Desktop\grbl-master\grbl\examples\grblUpload\grblUpload.ino:27:

C:\Users\Admin\Documents\Arduino\libraries\grbl/nuts_bolts.h:26:0: warning: "true" redefined

 #define true 1

 ^

In file included from C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:24:0,

                 from sketch\grblUpload.ino.cpp:1:

c:\program files (x86)\arduino\hardware\tools\avr\lib\gcc\avr\4.9.2\include\stdbool.h:43:0: note: this is the location of the previous definition

 #define true true

 ^

In file included from C:\Users\Admin\Documents\Arduino\libraries\grbl/grbl.h:43:0,

                 from C:\Users\Admin\Desktop\grbl-master\grbl\examples\grblUpload\grblUpload.ino:27:

C:\Users\Admin\Documents\Arduino\libraries\grbl/nuts_bolts.h:51:0: warning: "max" redefined

 #define max(a,b) (((a) > (b)) ? (a) : (b))

 ^

In file included from sketch\grblUpload.ino.cpp:1:0:

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:93:0: note: this is the location of the previous definition

 #define max(a,b) ((a)>(b)?(a):(b))

 ^

In file included from C:\Users\Admin\Documents\Arduino\libraries\grbl/grbl.h:43:0,

                 from C:\Users\Admin\Desktop\grbl-master\grbl\examples\grblUpload\grblUpload.ino:27:

C:\Users\Admin\Documents\Arduino\libraries\grbl/nuts_bolts.h:52:0: warning: "min" redefined

 #define min(a,b) (((a) < (b)) ? (a) : (b))

 ^

In file included from sketch\grblUpload.ino.cpp:1:0:

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:92:0: note: this is the location of the previous definition

 #define min(a,b) ((a)<(b)?(a):(b))

 ^

In file included from C:\Users\Admin\Documents\Arduino\libraries\grbl/grbl.h:43:0,

                 from C:\Users\Admin\Desktop\grbl-master\grbl\examples\grblUpload\grblUpload.ino:27:

C:\Users\Admin\Documents\Arduino\libraries\grbl/nuts_bolts.h:55:0: warning: "bit" redefined

 #define bit(n) (1 << n) 

 ^

In file included from sketch\grblUpload.ino.cpp:1:0:

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:123:0: note: this is the location of the previous definition

 #define bit(b) (1UL << (b))

 ^

X3msnake, 2018-01-11:
Make sure you dont have a old library installed too

mlo300, 2018-01-11:
when can search ultimate version?
tanks...

X3msnake, 2018-01-12:
https://github.com/gnea/grbl/wiki/Compiling-Grbl

J-Dunn, 2018-02-08:
I do not see any "compilation error" , they are WARNINGS. They will not prevent the compilation from finishing and producing an executable file. 

If the last thing shown is the size of the file etc. it worked. 
It is safe to ignore those particular warnings. You should learn the difference between a warning and an error and not report "errors" when you see warnings.

GaramMasala, 2018-02-11:
Arduino: 1.8.2 (FreeBSD), Board: "Arduino/Genuino Uno"

/usr/local/arduino/arduino-builder -dump-prefs -logger=machine -hardware /usr/local/arduino/hardware -tools /usr/local/arduino/tools-builder -built-in-libraries /usr/local/arduino/libraries -libraries /home/bernt/sketchbook/libraries -fqbn=arduino:avr:uno -ide-version=10802 -build-path /tmp/arduino_build_739072 -warnings=all -build-cache /tmp/arduino_cache_648643 -prefs=build.warn_data_percentage=75 -verbose /home/bernt/sketchbook/libraries/grbl/examples/grblUpload/grblUpload.ino
/usr/local/arduino/arduino-builder -compile -logger=machine -hardware /usr/local/arduino/hardware -tools /usr/local/arduino/tools-builder -built-in-libraries /usr/local/arduino/libraries -libraries /home/bernt/sketchbook/libraries -fqbn=arduino:avr:uno -ide-version=10802 -build-path /tmp/arduino_build_739072 -warnings=all -build-cache /tmp/arduino_cache_648643 -prefs=build.warn_data_percentage=75 -verbose /home/bernt/sketchbook/libraries/grbl/examples/grblUpload/grblUpload.ino
Using board 'uno' from platform in folder: /usr/local/arduino/hardware/arduino/avr
Using core 'arduino' from platform in folder: /usr/local/arduino/hardware/arduino/avr
Detecting libraries used...
"/usr/local/arduino/tools-builder/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/usr/local/arduino/hardware/arduino/avr/cores/arduino" "-I/usr/local/arduino/hardware/arduino/avr/variants/standard" "/tmp/arduino_build_739072/sketch/grblUpload.ino.cpp" -o "/dev/null"
Generating function prototypes...
"/usr/local/arduino/tools-builder/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR   "-I/usr/local/arduino/hardware/arduino/avr/cores/arduino" "-I/usr/local/arduino/hardware/arduino/avr/variants/standard" "/tmp/arduino_build_739072/sketch/grblUpload.ino.cpp" -o "/tmp/arduino_build_739072/preproc/ctags_target_for_gcc_minus_e.cpp"
/home/bernt/sketchbook/libraries/grbl/examples/grblUpload/grblUpload.ino:27:18: fatal error: grbl.h: No such file or directory
kompilering avslutad.
exit status 1
Error compiling for board Arduino/Genuino Uno.

Followed the instructions here
https://github.com/gnea/grbl/wiki/Compiling-Grbl