grbl/grbl #1340: Hi,

CNCLover, 2018-01-13:


CNCLover, 2018-01-13:
HI,
I am new to this software.
I have two uni-polar stepper motor and a bi-polar stepper motor.
Also have two KID65003AP drivers. for uni-polar stepper motors and will have two A4988 driver shortly.

I know that grbl is for uni-polar stepper motors. Not for both uni-polar and bi-polar.
If some one has finished this kind of job then I am interested to know how!

Best Regards.

radiolee, 2018-01-13:
leave the center of uni-polar NC just ok.

CNCLover, 2018-01-13:
Hi,
#
What about KID65003AP driver ic ? I must have to use one of them.

Best Regards.

X3msnake, 2018-01-13:
You would need a unipolar driver that adhered to the pololu pinout standart

http://www.ti.com/general/docs/lit/getliterature.tsp?baseLiteratureNumber=TIDUBB3&fileType=pdf

Easy way is to just use it with a bipolar stepper driver like it was said

X3msnake, 2018-01-13:
> baseLiteratureNumber=TIDUBB3&fileType=pdf

You could eventually use it with a driver like this

https://www.google.pt/url?sa=t&source=web&rct=j&url=http://www.produktinfo.conrad.com/datenblaetter/1200000-1299999/001261448-da-01-en-IC_MOTOR_DRIVER_DRV8805PWP_TSSOP_16_TID.pdf&ved=2ahUKEwik1u6qztXYAhUKchQKHQJoAJI4ChAWMAl6BAgNEAE&usg=AOvVaw3FKpS_p5qjhmRsNKMG1FBX

No dia sábado, 13 de janeiro de 2018, Vinicius Silva <x3msnake@gmail.com>

X3msnake, 2018-01-13:
> ww.produktinfo.conrad.com/datenblaetter/1200000-1299999/0012
> 61448-da-01-en-IC_MOTOR_DRIVER_DRV8805PWP_TSSOP_16_TID.pdf&
> ved=2ahUKEwik1u6qztXYAhUKchQKHQJoAJI4ChAWMAl6BAgNEAE&usg=AOv
> Vaw3FKpS_p5qjhmRsNKMG1FBX

This easy driver manual shows how to connect a 6wire unipolar motor to a

6-Wire Unipolar Stepper Motor Control Using The Big Easy Driver

https://www.google.pt/url?sa=t&source=web&rct=j&url=http://www.egr.msu.edu/classes/ece480/capstone/spring12/group03/Documents/Austin_ApplicationNote.pdf&ved=0ahUKEwjYkeC_0dXYAhWMWBQKHYPiAxYQFghAMAA&usg=AOvVaw1ESkdpbsbzMr11P3UBRH5f

No dia sábado, 13 de janeiro de 2018, Vinicius Silva <x3msnake@gmail.com>

closed 2018-01-14 by CNCLover