grbl/grbl #1341: Mirrored XYZ axis, probing

ApplePie420, 2018-01-13:
Hi,

i bought CNC 2418 not long ago. It came with pre-loaded 0.8 grbl firmware along with grblControl firmware. I was succesfully able to do some testing as well as PCBs, but i run into problem with autoleveling. As i knew, grbl 0.8 and its software does not support autolevelling, so i flashed my CNC with newest 1.1 version of firmware and downloaded Candle software. Since then, i got every axis mirrored, e.g. when i press right button on jog, it goes left, and vice versa for all of axises. And i also want to ask, how autolevelling works and how to do this, because i am unable to find any resources online for more than a week. Thanks

X3msnake, 2018-01-13:
Just use a sender with autoleveller included like ugs ou bcnc

As for the axis you need to use the invert the mask on the settings

This is all well documented on grbl wiki

https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration

ApplePie420, 2018-01-14:
Ok, thank you very much.
I was able to reverse the YZ axis, as only those were flipped. And i also managed to probe my PCB using Candle firmware, BUT i have a problem here. When i probe it, and apply the height map, it scales down. I have 1.5mm thick trace, but i cuts it all, and its only aroud 0.8mm wide. Also, the standart cutout for resistor is (in diameter) as big as resistor´s lead itself. Any suggestions?
Thanks

X3msnake, 2018-01-14:
Can you share some photos.
The endmill you are using, a printscreen of the board and a photo of the result?

What are you using to make the g-code?

Either you are using too big of a mill, you have bad settings on the toolpath, bad origin zero or simply just too much z depth

ApplePie420, 2018-01-20:
[image]
Ok..

I am using Eagle/EasyEDA to make layout, then FlatCAM to export gerber to g-code and Candle to mill.
The bit is V shaped 30° mill. The depth problem was solved, but the scailing is still the case, as i described before.

[image]

X3msnake, 2018-01-23:
are you originally working in mm or inches?

ApplePie420, 2018-01-25:
millimeters..

ApplePie420, 2018-01-25:
Now it went completly crazy.. I´ve still got scaling problem, which i am absolutely mad about. both FlatCAM and Candle are set to mm, but i still got like 13mm board from original 51mm...

J-Dunn, 2018-02-08:
If you have scaling issues it would seem that you don't have the correct axis parameters. ie steps per mm : $100 $101 and the depth issue is presumably $102 scaling. 

Check steps per rotation on your stepper motors ( any gearing you have ) and the lead screw pitch. Ensure that these combine to give the correct steps per mm values set up in  GRBL.  You also need to account for whether you are using full-step, half-step or mirco-step modes in your stepper drivers. 

For example 1.8 degree motors ( 200 steps per rev ) and 5mm pitch screws in half step mode will give 80 steps/mm.  200*2/5

ApplePie420, 2018-02-19:
well thats nice.. where i can get those parameters? i´ve got mine CNC from china and its really poor documented, and if then in chinese. I am not able to figure out anything, sorry :/ any recommendation?

J-Dunn, 2018-02-19:
work them out yourself. How many steps per rotation; what is the pitch of the ballscrew thread ; then work out steps per mm and correct your GRBL params.

chamnit, 2018-02-19:
Try the Grbl [Wiki](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration).

closed 2018-02-19 by chamnit

ApplePie420, 2018-02-22:
h