grbl/grbl #1345: help with GRBL unipolar motor fork

wydmynd, 2018-01-22:
I am using a[ fork of GRBL](https://github.com/ruizivo/GRBL-28byj-48-Servo/commits/master/grbl) that adds support for unipolar motors (only cost effective option for educational robotic kits in my country) 
unfortunately I designed my cnc as corexy configuration, and enabling this option does not affect behavior. 
I cannot understand if it is possible to make corexy kinematics work with this unipolar hack. the corexy code is mainly in system.c and the unipolar hack is done mainly in stepper.c 
it is a bit complex for me to understand how they connect.
much thanks for any tips.

X3msnake, 2018-01-22:
Can't you use a regular bipolar driver and connect the unipolar motors as
bipolar?

http://www.instructables.com/id/Running-a-Unipolar-Stepper-Motor-with-a-Bipolar-Dr/

http://www.ti.com/lit/ug/tidubb3/tidubb3.pdf

https://www.youtube.com/watch?v=yasCYgzNKwU

reallycurious, 2018-05-20:
Tested the above mentioned [library](https://github.com/ruizivo/GRBL-28byj-48-Servo) with COREXY configuration. It works as intended. Motor connections are as explained in "README.md" D2, D3, D4, D5 for stepper A and A0, A1, A2, A3 for stepper B. 
The only problem I encounterd while compiling the library is compiler warns and refuses library when COREXY is enabled as compiled file is too large for atmega328. A goofgle search pointed me to https://discuss.inventables.com/t/inventables-fork-of-grbl-too-large-32-266-bytes-when-compiled/28148 which explained why. After downgrading the Arduino AVR Boards to 1.6.11 through Board manager the library compiled successfully.