grbl/grbl #1348: grbl 0.8  Please fix the error message.

yunocho, 2018-01-30:
$$ (view Grbl settings)
$# (view # parameters)
$G (view parser state)
$I (view build info)
$N (view startup blocks)
$x=value (save Grbl setting)
$Nx=line (save startup block)
$C (check gcode mode)
$X (kill alarm lock)
$H (run homing cycle)
~ (cycle start)
! (feed hold)
? (current status)
ctrl-x (reset Grbl)

If you hit the above command, an error message will be displayed.

[image]

I get an error message like the picture below. Is there a solution?

X3msnake, 2018-01-30:
Any special reason for using 0.8 still?

what sender are you using?

109JB, 2018-01-31:
The error messages you show on the picture are garbled v1.1 error messages.

Error 5 means homing isn't configured
Error 8 means grbl isn't idle when you issue a $ command