grbl/grbl #1350: stuck in steps

CNCLover, 2018-01-30:
HI,
I stuck in steps configuration.

I am using a4988, 48 steps stepper motor.
Grbl firmware and universal g code sender.
I haven't find where to select steps of stepper motor.
I am new to this so requesting help.
Any link related to it will also be good.

Thanks

X3msnake, 2018-01-30:
Check the official wiki

$100=250.000 X steps/mm
<https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#100-101-and-102--xyz-stepsmm>
$101=250.000 Y steps/mm
<https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#100-101-and-102--xyz-stepsmm>
$102=250.000 Z steps/mm
<https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#100-101-and-102--xyz-stepsmm>
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration

Use the reprap calculator to see how many steps you must declare

https://www.prusaprinters.org/calculator/#steppermotors

Choose the calculator according to you mechanical setup (belt or screw)

jahnj0584, 2018-01-30:
This is better.

https://nathan7.eu/stuff/RepRapCalculator/RepRapCalculator.html#MotorStuffPrecise

X3msnake, 2018-01-30:
@jahnj0584 

Didn't knew that one, thanks for sharing :+1: 

Funny how that there's no calculators dedicated to CNC only :)

jahnj0584, 2018-02-01:
Winder has it noted to eventually add this "calculator" into UGS natively
for new machine set up. For now my link is the best i found.

CNCLover, 2018-02-05:
I am building 3d printer and pick and place machine.
Now I feels like i'm not good in software side.

Any suggestions?

CNCLover, 2018-02-05:
Hi,
Can I use GRBL for 3d printer?
I am facing problems with pronterun.
[image]
GRBl is working best with universal g code sender. Should I use grbl with it?

Thanks.

J-Dunn, 2018-02-08:
Marlin is the classic 3d printer firmware.  It is historically derived from GRBL. Is there are reason you are not using that?

CNCLover, 2018-02-11:
Marlin's firmware is not supporting arduino nano based board. So I now installed teacup for arduino nano.

Looks like 3d printer's software part is almost done.

Focus at 6:45

https://www.youtube.com/watch?v=i0ZT4y3DYrY&index=7&list=LLc9AOgrVKEyt000eWQ5hbhw&t=59s

He used 3d printer software for pick and place machine.
Which way will be more convenient?
A)  I note down Gcodes as per my requirements into a file and then upload to grbl firmware 
B)  Use 3d printer software like this man?

Thanks

J-Dunn, 2018-02-11:
3d printer and PnP m/c will require 4th axis not available in Uno/GRBL.  Why don't you buy hardware that is suitable since you seem rather weak on software and not likely to modify GRBL. 

If you buy an Arduino Mega based board you will have what you need and can use std solutions, like Pronterface and Marlin.

https://fr.aliexpress.com/item/3Dpriter-control-board-MKS-SBase-V1-0-32-s-Motherboard-compatible-Smoothieware-open-source-firmware-support/32384368763.html?spm=a2g0w.10010108.1000016.1.24011dddkkoTfJ&isOrigTitle=true

CNCLover, 2018-02-11:
It's not good reason to buying PNP machine because it's cost will be higher than the main project. So not suitable to invest/buy. For such low quantity, PCB assemblers are also not ready to deal with me.  

4th axis? Rotary motion? I don't need to rotate components. I will position them as per requirement. 0° and 90° .

I still prefer to make own PNP machine.

If you know Gcodes that would help me in this, just post them. I will learn them in detail in wiki. I have searched for it and there were so many. So feeling lots of learning new codes! And many of them are required for video help to understand operations. 

Do you know simple digital high/low  can be controlled with grbl?

Thanks

J-Dunn, 2018-02-11:
4th axis? Rotary motion? 

No, look at the guy's video, he has a stepper on the vacuum pump for picking the pieces.  On a 3D you will need another "axis" for the extruder. 

> Do you know simple digital high/low can be controlled with grbl?

No , seriously,  if that is the kind of question you are needing to ask you are way out of your depth, and no one here is going to walk you through a person tuition . 

I suggest you go over to arduino.cc and play around with a few more basic projects and learn you way around before trying to do an advanced project.

X3msnake, 2018-02-11:
if you want a 3dprinter/cnc you need a mega and try repetier as firmware

they have a config tool https://www.repetier.com/firmware/v100/index.php

CNCLover, 2018-02-12:
J-Dunn, I don't think that that guy used stepper motor for vacuum purpose. It can done with simple motor. Only required to turn on-off. That's why I mentioned " Do you know simple digital high/low can be controlled with grbl?".
If cnc machine's rotoy drill (not perfect work) like x-curve machine can be controlled, then why same vacvum pump can't be operated?

If you want to understand my hardware, I can post here. It's simple.

Thanks

X3msnake, 2018-02-12:
use the spindle pin on/off  with a power transistor and add a solenoid
magnet or a solenoide valve

J-Dunn, 2018-02-12:
why don't you just do basic research instead of asking questions?  GRBL/Uno does not do SD. The card I previously linked does.

CNCLover, 2018-02-12:
Hmm, ok. Thanks

closed 2018-02-12 by CNCLover

X3msnake, 2018-02-12:
there is a forked project that uses a second uno to do SD card and LCD

have a look at the GRBL wiki

https://github.com/gnea/grbl/wiki/Frequently-Asked-Questions#extending-grbl