grbl/grbl #1353: [Q] Has Grbl fixed last-segment chaining?

thinkyhead, 2018-02-05:
Hey Grblers, I've been tracking a bug in Marlin Firmware (based on an older version of Grbl) where the last segment of any move is not chained to the previous one. For example, this G-code:

```gcode
G28
G1 Z5

G1 X50 F4500
G1 X75
G1 X100
M400
G1 Y50
G1 Y75
G1 Y100
M400
G1 X50
G1 X25
G1 X0
M400
G1 Y50
G1 Y25
G1 Y0
```
…should behave identically to this G-code:
```gcode
G28
G1 Z5

G1 X100 F4500
M400
G1 Y100
M400
G1 X0
M400
G1 Y0
```
However, there are pauses before the last segment on each side of the square.

Was this a known issue with Grbl around 2012 or so, which has since been fixed?

chamnit, 2018-02-05:
Not since 2011, since I started working on this project. It was one of the first things I fixed. Marlin was forked right before then.

closed 2018-02-05 by chamnit

thinkyhead, 2018-02-05:
You wouldn't happen to have a commit id, would you? If not, I'm happy to search the blame logs.

thinkyhead, 2018-02-05:
This looks like a potential candidate… https://github.com/grbl/grbl/commit/d75ad82e4932aea166c189e4160e197e4710c191#diff-a4eaa8474f0ada716c21c0d502f87a6dR132
```diff
-  planner_reverse_pass_kernel(NULL, block[0], block[1]);
+  // Skip buffer tail to prevent over-writing the initial entry speed.
```

thinkyhead, 2018-02-08:
I'm going out on a limb and betting grbl still has this issue. After much studying of the code, without guidance, I cannot see why it would not. Please test the posted G-code and see if it performs correctly.

chamnit, 2018-02-08:
@thinkyhead : Nope. Grbl runs just fine. There hasn't been a significant motion issue in several years, which is the main reason why its so popular, even though its restrictive and simple. What you are likely seeing is a buffer starvation or inability to plan the path quickly enough and causing a desync between execution and the planner. The fixes in Grbl are subtle, but are exact. They account for failure conditions gracefully.

thinkyhead, 2018-02-25:
Thanks much. A lot to digest in studying the updates to your implementation.