grbl/grbl #1355: D11 goes HIGH on RESET, causing Laser to turn on

gamerpaddy, 2018-02-13:
I got a laser on my 3 axis router, connected to Pin 11 for laser mode. But when i connect to my arduino or reset the board, D11 turns on and applies full Power to the Laser.  

Is there any fix for that without using a physical interlock?

X3msnake, 2018-02-13:
"Be very careful. I've seen the laser fire while the arduino is booting.
Make sure to connect a 10K resistor from Arduino ground to LPS TH. The
arduino boots with all pins as inputs and pin 11 can float up enough to
activate the laser."

https://github.com/grbl/grbl/issues/1263

gamerpaddy, 2018-02-13:
Hey thanks for the tip. I didnt know the Arduino is doing that, i knew its pulling the pin to either high or low. But at the input setting its at HighZ, floating.  
I tried with a 10k between TTL and GND on my laser driver and when i now connect to my arduino, the laser flashes briefly. much less power behind than S1 (from 255).  Im now using a 4.7k and it stays off until i M3 it.  
Thanks.

X3msnake, 2018-02-13:
You are welcome

00alkskodi00, 2018-06-13:
Hi 

i am novice

can that working?

[image]