grbl/grbl #1356: 5:1 scale movements?

dvanc, 2018-02-15:
Very new to all of this, but recently bought a chinese CNC and had problems from the outset. Assembly went fine, but the programs would crash part way through. Figured out it was most likely the board, so I bought an UNO and CNC Shield. Finally got it all set up and working, but the scale is off.  A request for a 10mm movement along any axis yields a 50 mm movement. I tried gbrl .9 and 1.1 with the same result. I don't even know where to start. Is it a setting in the code I can change?

langwadt, 2018-02-15:
eeprom settings, https://github.com/grbl/grbl/wiki/Configuring-Grbl-v0.9#100-101-and-102--xyz-stepsmm

jahnj0584, 2018-02-15:
Calibrate your axis
https://nathan7.eu/stuff/RepRapCalculator/RepRapCalculator.html#MotorStuffPrecise

100-101-and-102--xyz-stepsmm

<https://github.com/grbl/grbl/issues/1356#issuecomment-366099894>, or mute
the thread

dvanc, 2018-02-15:
Thank You! This has been a bit of a steep learning curve, I appreciate the help!