grbl/grbl #1357: universial controls......

drbob2456, 2018-02-17:
I have download every universal controller programs. not a one seems to work right. now my z axis is frozen. I need help!

X3msnake, 2018-02-17:
Mmm, some more info would be good

What is your GRBL version

What is your machine (custom, bought)

What is your hardware (DIY Breakoutboard wuth uno, china breakoutboard
version x, other)

X3msnake, 2018-02-17:
Please describe what do you mean by not working right

Please describe what do you mean by Z axis frozen

2018-02-17 19:15 GMT+00:00 Vinicius Silva <x3msnake@gmail.com>:

> Mmm, some more info would be good
>
> What is your GRBL version
>
> What is your machine (custom, bought)
>
> What is your hardware (DIY Breakoutboard wuth uno, china breakoutboard
> version x, other)
>
>
>> I have download every universal controller programs. not a one seems to
>> work right. now my z axis is frozen. I need help!
>>
>>
>
>
>
> Com os melhores cumprimentos,
> Vinicius Silva

drbob2456, 2018-02-17:
Grbl controller 3.6.1

X3msnake, 2018-02-17:
That is the sender you are trying to use I mean the firmware version of
your board.