grbl/grbl #1359: Machine and work positions

Janernevis, 2018-02-23:
Hi to everyone!
I am new in the world of CNCs so be gentle with me. I got myself CNC machine and I am still learning.
My machine is based on Arduino Uno and CNC Shield with A4988 stepper drivers. As front end app I am using UGS platform.
I have a question about work and machine positions shown at UGS platform (latest version) which puzzles me a bit and guys from UGS directed me to this place coz they said it's grbl related.

I will try to explain to you as good as I can.
Let's say I have machine with dimensions 500x500mm and my spindle is in top right corner and that position is marked as machine position X0 Y0 Z0 and work position X0 Y0 Z0. If I go 300 down and 300 left those positions will change to X-300 Y-300 Z0. If I then click on Reset zero button, work positions will be X0 Y0 Z0 and machine position will stay X-300 Y-300 Z0. If I disconnect connection and connect again my work positions will be X300 Y300 Z0 (opposite value than before) and machine position will be X0 Y0 Z0. If I then click on Return to zero button my spindle will go down and left (instead of up and right) and hit the edge because it will actually go to X-600 Y-600 Z0.
I know I'm a newbie but that kind of behavior is so wrong to me. Can someone explain to me why is that programmed like that, am I dumb and don't understand it correctly or I did something wrong during setup or I hit some wrong button?
All I want is for machine to remember that top right corner is X0 Y0 Z0 and that it has space 500mm till bottom and 500mm till left edge. And while I'm working on some projects I can change work zero position for thousand times but when I disconnect connection, that machine remembers current position so that next time when I establish connection and give it command to go to machine zero it will know exactly where that is (considering that in meantime I haven't moved axis by force, by hand or something like that). Is that possible?

chamnit, 2018-02-23:
@Janernevis : Stepper motors are open-loop, meaning they have no reference point. Upon startup, Grbl doesn't and can't know where it is so it initializes as zero. You need homing switches that will provide you that reference point. See the wiki for information.

closed 2018-02-23 by chamnit

Janernevis, 2018-02-23:
Thank you on quick response.

Let's leave home switches a side for now, like they don't exist.

When I change position I can see that grbl registers that change. If I told grbl where my 0 point is and it registers every move I made, then obviously it knows where spindle is right now. As I said before, if I disconnect from my Arduino board and then again connect to it, it shows me my last coordinates, but in opposite values.
First, can you explain to me coz I am new and I don't know, why it shows opposite values than before disconnect? Why not exact coordinates as before?
Second, if it shows me values from last time then it knows where that point is in accordance to start 0 point, isn't it? And therefore it knows where every other point is and how much space it has till machine edges coz I entered machine dimensions in grbl settings.
Do you understand what I am saying? And can you explain to me where's the error in my logical thinking?
For me, it looks like grbl registers every move and saves last position before disconnect (because it gives me opposite values next time I connect) so that's in collision with what you are telling me. I want to understand.

Also, in this discussion we have to take in consideration that spindle hasn't been moved manually, by force, during the machine was power off.

langwadt, 2018-02-23:
if I understand you correctly I believe what you see is that the offset between work and machine coordinates is stored in eeprom when you reset work position to 0,0,0. i.e. wcs = mcs + offset, 
[0,0,0] = [-300,-300,0] + [300,300,0]

when you reset the arduino mcs is set to 0,0,0 that's all the machine knows
thus when applying the offset you get wcs = [0,0,0] + [300,300,0] = [300,300,0]

Janernevis, 2018-02-24:
OK, than you on that clarification.
But, what's the point of that? That doesn't give you any functionality.

If that would be setup as I wrote before, machine would always know where it is if it hasn't been moved by force during power off.

langwadt, 2018-02-24:
if you have limit switches and home the machine after power off, you are back to the same wcs. 
You can't assume anything if the power has been off

109JB, 2018-02-24:
> OK, than you on that clarification.
But, what's the point of that? That doesn't give you any functionality.

> If that would be setup as I wrote before, machine would always know where it is **_if it hasn't been moved by force during power off._**

You cannot assume that it has to be moved "by force".  The Z-axis on my CNC milling machine will slowly descend until it is touching the machine table, and this is not by force, but just under its own weight. An unpowered belt driven axis or rack and pinion can by moved by s light push in many cases.

As @langwadt said, you can't, or at least shouldn't assume the machine is in a certain position at startup.  If you do and your assumption is wrong, you run the risk of damaging the machine, or the parts you are making, or on a larger machine you could be risking injury.

chamnit, 2018-02-24:
What you are describing is closed loop control which involves way more stuff and makes it expensive. Stepper motors are open loop and just need to home to fixed switches. There are tons of resources that will explain this to you if you google it.

jahnj0584, 2018-02-24:
Also even servo machines need to be homed every power cycle... It's kinda
like complaining that every time you ride a bike you have to reset the
pedals to start going.
"well, i didn't touch the pedals since i got off so they have to be where i
left them"

Janernevis, 2018-02-25:
OK, guys. Thank you all on explanation.