grbl/grbl #1361: grbl with error the program stopped working

Dreamlikez, 2018-03-01:
[image]
I'm a beginner and I started recently.
I use grbl version 0.8 and some random moments the program stops working and many bugs appear
the program closes from 2 to 10 seconds after being opened randomly, I can move the motors in this period
I'm researching but I do not know what might be causing this

X3msnake, 2018-03-01:
upgrade to 1.1 and use one of the listed Senders in the wiki

https://github.com/gnea/grbl/wiki/Using-Grbl

X3msnake, 2018-03-01:
but before connect by serial to your machine using arduino IDE for example
and run settings $ on the terminal window

Copy and paste you machine settings to a txt file in case the machine was
not build from scratch by you.

If you are using 0.8 you are probably using a older breakout board and you
will have to change pin mapping before flashing 1.1 or change your wiring

PIN MAP GRBL 0.8 no pwm spindle
https://github.com/grbl/grbl/wiki/Connecting-Grbl

PIN MAP GRBL 1.1 Default
https://github.com/gnea/grbl/wiki/Connecting-Grbl (pin 11 changed with 12)

2018-03-01 19:25 GMT+00:00 Vinicius Silva <x3msnake@gmail.com>:

> upgrade to 1.1 and use one of the listed Senders in the wiki
>
> https://github.com/gnea/grbl/wiki/Using-Grbl
>
>
>
>> [image: forum]
>> <https://user-images.githubusercontent.com/36965201/36858470-01f1ad26-1d5a-11e8-828e-dfc5d480af28.png>
>> I'm a beginner and I started recently.
>> I use grbl version 0.8 and some random moments the program stops working
>> and many bugs appear
>> the program closes from 2 to 10 seconds after being opened randomly, I
>> can move the motors in this period
>> I'm researching but I do not know what might be causing this
>>
>>
>
>
>
> Com os melhores cumprimentos,
> Vinicius Silva

X3msnake, 2018-03-01:
To change the pin layout on the firmware side, use the Arduino Upload
method > https://github.com/gnea/grbl/wiki/Compiling-Grbl

edit config.h and comment out  define VARIABLE_SPINDLE

'''
// Enables variable spindle output voltage for different RPM values. On the
Arduino Uno, the spindle // enable pin will output 5V for maximum RPM with
256 intermediate levels and 0V when disabled.
// NOTE: IMPORTANT for Arduino Unos! When enabled, the Z-limit pin D11 and
spindle enable pin D12 switch!
// The hardware PWM output on pin D11 is required for variable spindle
output voltages.

#define VARIABLE_SPINDLE // Default enabled. Comment to disable.
'''

2018-03-01 19:38 GMT+00:00 Vinicius Silva <x3msnake@gmail.com>:

> but before connect by serial to your machine using arduino IDE for example
> and run settings $ on the terminal window
>
> Copy and paste you machine settings to a txt file in case the machine was
> not build from scratch by you.
>
>
> If you are using 0.8 you are probably using a older breakout board and you
> will have to change pin mapping before flashing 1.1 or change your wiring
>
>
> PIN MAP GRBL 0.8 no pwm spindle
> https://github.com/grbl/grbl/wiki/Connecting-Grbl
>
>
>
> PIN MAP GRBL 1.1 Default
> https://github.com/gnea/grbl/wiki/Connecting-Grbl (pin 11 changed with 12)
>
>
>
>
>
>
>
>
>
>
>
>
>
> 2018-03-01 19:25 GMT+00:00 Vinicius Silva <x3msnake@gmail.com>:
>
>> upgrade to 1.1 and use one of the listed Senders in the wiki
>>
>> https://github.com/gnea/grbl/wiki/Using-Grbl
>>
>>
>>
>>> [image: forum]
>>> <https://user-images.githubusercontent.com/36965201/36858470-01f1ad26-1d5a-11e8-828e-dfc5d480af28.png>
>>> I'm a beginner and I started recently.
>>> I use grbl version 0.8 and some random moments the program stops working
>>> and many bugs appear
>>> the program closes from 2 to 10 seconds after being opened randomly, I
>>> can move the motors in this period
>>> I'm researching but I do not know what might be causing this
>>>
>>>
>>
>>
>>
>> Com os melhores cumprimentos,
>> Vinicius Silva
>>
>
>
>
> Com os melhores cumprimentos,
> Vinicius Silva

Dreamlikez, 2018-03-01:
Yes, I am using a board ready with cnc shields. a board called woodpecker it came with that 0.8 software. I'm starting to learn about cnc now so I thought it would be better for me.

X3msnake, 2018-03-01:
Update to 1.1 following the instruction i gave you before.

Don't forget to connect via serial terminal first and save your machine
settings.

Or try a older version of the sender, look for a version build on the same
year as 0.8c

Dreamlikez, 2018-03-01:
I'm first tempted to 0.8c then I upgrade to 1.1 (because I bought the woodpecker board with that version inside it and I'm afraid to upgrade too) if it does not work.
I searched the site and did not find for a version 0.8c to download

Dreamlikez, 2018-03-01:
but can I update the software of a board already mounted? like this 
[image]

X3msnake, 2018-03-01:
Yes you can

This board uses arduino it is the same

Just plug the usb to the pc.

Btw Post here the settings

X3msnake, 2018-03-01:
If thephoto you posted is from you board it is supposed to come with grbl
0.9 and is compatible with 1.1

https://youtu.be/kJRy3_heliQ

X3msnake, 2018-03-02:
https://github.com/gnea/grbl/issues/336#issuecomment-369894707

Dreamlikez, 2018-03-04:
Thanks! I will try now