grbl/grbl #1362: One motor not spinning

skfahad, 2018-03-04:
I have made my CNC machine from but have run into a strange issue. To test the motor, I sent the command X10 Y10 from grbl controller, the problem is X axis motor running properly but Y axis motor not spinning in either direction. The strange thing is when I sent command through G Code generated by Makercam all the axis spinning.

chamnit, 2018-03-04:
By default, gcode is defined in absolute coordinates, not incremental.

skfahad, 2018-03-04:
I know it well. My question is why one axis not moving on X10 Y10 command in either direction but other axis working fine?

jahnj0584, 2018-03-04:
Use the jog command to test. If you are in absolute and already sit at Y10
nothing will move.

<https://github.com/grbl/grbl/issues/1362#issuecomment-370247119>, or mute
the thread