grbl/grbl #1368: Issues with 1.1f

epif18, 2018-03-18:
I loaded the latest grbl via the arduino IDE as well as flashed the .hex file with the same result.  1.1f20170801.  I have a woodpecker CNC board on a Linksprite.  I've tried selecting the UNO and Nano processor, both worked.  Compiling the source code resulted in a bunch of compile warnings but it loaded. I can jog x,y,z using the control panel.  I executed the $I command via the grbl control panel on my PC (Windows 10) and saw that 1.1f was loaded into the board.  The problem is that the $I command executes when I hit the send button but the control panel buttons seem to be sending the command forever (but it isn't) and the abort button will not abort nor will the pause button pause.  I can't abort a program either nor can I jog anything.  I have to hit the reset on the woodpecker board.  Is there a bug in 1.1f? There must be something wrong with my setup somewhere.

chamnit, 2018-03-18:
There is a minor EEPROM bug that is related to clearing the EEPROM for the new version. It won’t be fixed because it’ll cause way more problems to do so. It’s likely you may need to wipe your EEPROM with a $RST=* command.

epif18, 2018-03-18:
Thanks Sonny for the quick response.  I’ve looked at other Gcode senders and I don’t have the same problems with Candle (which has a GUI almost identical to grblcontrol which exhibits the problem) or bCNC.  I’ll keep you guys informed. I just installed limit switches and I’ll be testing the limits and Home functions next.

Mark

chamnit, 2018-03-18:
You have to use a GUI that has been updated for Grbl v1.1. There was some tweaks to the streaming protocol to include switch and realtime override data, along with some other minor stuff. It's likely you need to update your sender or switch to a v1.1 compatible one.

closed 2018-03-18 by chamnit