grbl/grbl #1372: GRBL 0.8 and CNC Shield 3.0

TrueCalfa, 2018-03-25:
Hello everyone,

On the last days I've been trying to build a CNC machine using old DVD Players, and Arduino UNO (Chinese Clone) and a Protoneer CNC Shield V3.0. I also used a polymeter to make sure i was wiring the stepper correctly.

Once i got my hardware ready i uploaded GRBL 1.1 to my arduino board and tried both GcodeSender and GRBL Controller, but didn't get any stepper to move. Then i read that CNC Shield v3.0 must be used with GRBL 0.8, so i downloaded XLoader and flashed GRBL 0.8 hex file to my arduino but again, didn't get my steppers to move. I tried to connect the wires in different ways too.

By the way, I'm using an old PC power supply to feed the shield, and made sure it's getting 12V.

I'm pretty newbie to everything regarding CNC so i don't really know what to do now. Can you guys help me, please?

Thanks a lot in advance for any help!!

X3msnake, 2018-03-25:
did you took note of the settings $$ that worked?

TrueCalfa, 2018-03-26:
Hi @X3msnake, i don't really understand what you mean, but i couldn't move the steppers a single step.

Minutes ago, tho, i have manually applied 12V to each of the coils and noticed it was moving a single step.

X3msnake, 2018-03-26:
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration

The settings is what define the motor move.

TrueCalfa, 2018-03-26:
Hi again,

This is what i get once i open GRBL Controller on COM10:
[image]

This is what i get once i type "$":
[image]

And, this is what i get when i type "$$":
[image]

Again, excuse me if this is not what you were asking me for. I feel totally lost :/

X3msnake, 2018-03-26:
Exactly that.

 did you take these values when it was working?

X3msnake, 2018-03-26:
Sorry i misread. You never had it run sorry.

Are you sure your motors are 12v? You should check the product number and
see what is the motor specs.

Namely the step angle and the max velocity

X3msnake, 2018-03-26:
try these settings

GRBL Settings $100=5.000 (x, step/mm) $101=5.000 (y, step/mm) $130=40.000
(x max travel, mm) $131=40.000 (y max travel, mm) Toggle this to
Enable/Disable Hard limit $21=0 (hard limits, bool)

Source: https://www.youtube.com/watch?v=kcz1ygh20c0

2018-03-26 14:04 GMT+01:00 Vinicius Silva <x3msnake@gmail.com>:

> Sorry i misread. You never had it run sorry.
>
> Are you sure your motors are 12v? You should check the product number and
> see what is the motor specs.
>
> Namely the step angle and the max velocity
>
> No dia segunda-feira, 26 de março de 2018, Vinicius Silva <
>
>> Exactly that.
>>
>>  did you take these values when it was working?
>>
>> No dia segunda-feira, 26 de março de 2018, TrueCalfa <
>>
>>> Hi again,
>>>
>>> This is what i get once i open GRBL Controller on COM10:
>>> [image: connect]
>>> <https://user-images.githubusercontent.com/28894098/37899567-335062e0-30ec-11e8-9e22-240d1cd3092b.jpg>
>>>
>>> This is what i get once i type "$":
>>> [image: default]
>>> <https://user-images.githubusercontent.com/28894098/37899650-6c71abb0-30ec-11e8-9497-1faf55d9f407.jpg>
>>>
>>> And, this is what i get when i type "$$":
>>> [image: default]
>>> <https://user-images.githubusercontent.com/28894098/37899786-c2412c64-30ec-11e8-903a-3519f713555f.jpg>
>>>
>>> Again, excuse me if this is not what you were asking me for. I feel
>>> totally lost :/
>>>
>>> <https://github.com/grbl/grbl/issues/1372#issuecomment-376112703>, or mute
>>> the thread
>>>
>>
>>
>> Com os melhores cumprimentos,
>> Vinicius Silva
>>
>>
>
> Com os melhores cumprimentos,
> Vinicius Silva

X3msnake, 2018-03-26:
Fuse? By polymeter i believe you are trying to say multimeter? If not I don't know what a Polymeter is.

You should always use the same fuse your equipment comes with otherwise the next time you blow something up might not be as cheap as the fuse ;)

TrueCalfa, 2018-03-26:
Yes, you are right, i meant multimeter.