grbl/grbl #1373: Wait after Spindle start for RPM to reach?

Ravenbs, 2018-03-26:
Hello,

When grbl start the spindle with some RPM value, it starts immediatly cutting.
But my spindle need 2-3 seconds to ramp up to this rpm value.
Think this is true for most of the spindles.

Would be great if there is a value in the config for "Wait x seconds after each speed change".

Or do I need to use dwell in post processor?
But the fusion 360 Post processor does not include this dwell as well.https://forums.autodesk.com/t5/fusion-360-computer-aided/adding-dwell-after-spindle-start/td-p/6229681

jahnj0584, 2018-03-26:
Either add a G4 P2 command to your post or use UGS which includes a dwell
on M3

Ravenbs, 2018-03-26:
What is UGS?

X3msnake, 2018-03-26:
Universal Gcode Sender

tkms002, 2019-05-16:
how do you enable the dwell in UGS?