grbl/grbl #1375: feed rate in Firmware & G-code

sandy9159, 2018-03-28:
I have one query
why is there feed rate are mention in G-CODE though feed rate are already set via firmware.
means 
when I am generating Gcode for pen plotter i have option to choose feed rate of my choice .

what if I choose feed rate higher or lower then what is set in firmware.... ?

X3msnake, 2018-03-28:
You really need to go learn some g-code :)

https://github.com/gnea/grbl/wiki/Frequently-Asked-Questions#g-codes

http://linuxcnc.org/docs/html/gcode.html

http://www.linuxcnc.org/docs/2.4/html/gcode_overview.html

But here's the short answer.
Firmware sets your machine max feed rate

G-code allows you to use anything up to that. If you set above the max it
will use the max instead

sandy9159, 2018-03-28:
Thanks for you guide now it make sense for me