grbl/grbl #1379: Vexta 5 Phase Stepper with Driver for GBRL

rickysupreme, 2018-04-02:
I have the original vexta stepper 5 phase 0.72degrees with the stepper driver http://www.orientalmotor.com/products/pdfs/opmanuals/HM-60042E.pdf It is the driver without the directional switch pins. I have found a basic arduino sketch that will drive it. I basically just need pulses to one pin to drive it clockwise and pulses to drive another pin to turn it counter clockwise. I want to drive it with nano dlp so i need it to be able to interpret G code and therefore I need someone to modify GBRL to send pulses to either the clockwise or counter clockwise pins of my stepper driver. I do not know how to modify GBRL or any other complicated code for that matter, I will donate if you can help.

langwadt, 2018-04-02:
it is simple enough to modify if you know a bit of programming, but I think you could use grbl as-is by adding a transistor and a resistor

X3msnake, 2018-04-02:
Your best approach is going hardware mode with the trasistor like @langwadt
said. you do not have free pins left for the required changes on the nano
or the uno.

Basically you have to make or buy a converter circuit that converts
STEP/DIR to CW or CCW

Like this
http://www.electronics-lab.com/project/stepdir-signal-cwcww-signal-converter-cnc-motion-control-systems/

langwadt, 2018-04-02:
since the drive has optocoupler inputs this might do, 

[image]

X3msnake, 2018-04-02:
@langwadt

Cool, that's quite easier to implement :)

Yet you will be feeding 5V to the CCW- when you are pulling CW- to ground
won't that be bad for the driver?

X3msnake, 2018-04-02:
Shouldn't he add a diode to avoid reverse polarity?

2018-04-02 18:56 GMT+01:00 Vinicius Silva <x3msnake@gmail.com>:

> @langwadt
>
> Cool, that's quite easier to implement :)
>
> Yet you will be feeding 5V to the CCW- when you are pulling CW- to ground
> won't that be bad for the driver?
>
>
>
>
>> since the drive has optocoupler inputs this might do,
>>
>> [image: 2018-04-02 19 38 35]
>> <https://user-images.githubusercontent.com/1613189/38207650-e33832a6-36ae-11e8-9d0f-ba2351798e18.jpg>
>>
>> <https://github.com/grbl/grbl/issues/1379#issuecomment-377991935>, or mute
>> the thread
>>
>
>
>
> Com os melhores cumprimentos,
> Vinicius Silva

langwadt, 2018-04-02:
you are right a diode on CCW- is needed, because the Arduino probably doesn't have enough drive  to drive the extra ~20mA through the reverse polarity diode on the CCW input

cri-s, 2018-04-02:
Using optocouplers usually it is enought to connect cw+ to ccw- and cw- to
ccw+ connecting cw+ to step and ccw+ to dir.
Naturally the Dir/step delay compile time option  is necessary.

langwadt, 2018-04-02:
@cri-s because of the reverse diode in the driver the reversed input will draw ~20mA at 5V that is a bit much for an arduino output, and the step polarity will change with direction I think that'll cause extra  steps

cri-s, 2018-04-02:
Good catch. Then Connect it serially.
I prefer a nand IC having Vih over 2v for the conversion that is connected
ON - and + ON 5v.

rickysupreme, 2018-04-03:
Can you please draw a circuit diagram if you think it will work? I can't quite follow otherwise.

rickysupreme, 2018-04-03:
Also I only need two input pins for the driver cw and ccw the others go to ground In the setup that worked with the arduino sketch.

langwadt, 2018-04-03:
@rickysupreme you only need one motor?

cri-s, 2018-04-03:
Here a circuit to convert step/dir to cw/ccw.
Use 74HC00 or 74AC(T)00 instead of the 74C00.
The transistor version works equals, the only esitation i have on the
transistor is the
low on threshold that eventually during back emf could be problematic
(in theory).

X3msnake, 2018-04-03:
> Who can check this for me. It actually worked but need to know if it could
> damage my driver or result in bad stepper accuracy.
>
> [image: unnamed 1]
> <https://user-images.githubusercontent.com/7353396/38254843-9a655a82-379d-11e8-806b-08fc374e80e3.jpg>

Looks ok

What is the resistor value you are using?

X3msnake, 2018-04-03:
LM35?

I can only see temperatur sensor datasheets, are you sure that's the
transistor you are using?

2018-04-03 16:21 GMT+01:00 Vinicius Silva <x3msnake@gmail.com>:

> Looks ok
>
>
> What is the resistor value you are using?
>
>
>> Who can check this for me. It actually worked but need to know if it
>> could damage my driver or result in bad stepper accuracy.
>>
>> [image: unnamed 1]
>> <https://user-images.githubusercontent.com/7353396/38254843-9a655a82-379d-11e8-806b-08fc374e80e3.jpg>
>>
>> <https://github.com/grbl/grbl/issues/1379#issuecomment-378266282>, or mute
>> the thread
>>
>
>
>
> Com os melhores cumprimentos,
> Vinicius Silva

rickysupreme, 2018-04-03:
Well its a real ordinary looking resistor about 4mm wide with the shape of a half cylinder.  the resistor seems to be marone, black, black, gold, gold. it is either 1k or 10k. this is all stuff from the fake arduino uno starter kit. The axis is working with exact millimeter precision even without the resistor. Not sure if the resistor is necessary at all, nothing seems warm.

X3msnake, 2018-04-03:
The resistor is there to protect the arduino pin ;)

As for the LM35 i believe it to be a temperature sensor but it does look
like it can be used as a transistor albeit it is not it's pourpose

https://www.element14.com/community/thread/36219/l/lm35-dz-same-as-pnp-transistor?displayFullThread=true

rickysupreme, 2018-04-03:
Thats pretty funny, Let me know what resistors and transistor I should use and you will also win a prize. I want to achieve less than 10 micron repeatability because I am an idealist.  langwadt is the first winner for suggesting this circuit.

rickysupreme, 2018-04-03:
cri-s can you also simplify your explanations somehow, maybe a diagram, you could still win the bonus prize after all. You seem to know what you are talking about.

langwadt, 2018-04-03:
[image]

or if you one need one motor and can compile the code there's a one line hack to do it in software

X3msnake, 2018-04-03:
Sorry i'm not good enough with electronics to advise you in what transistor
to use.

I guess any PNP transistor would work common pnp transistors are  2N3904
and 2N2222

As for the percision it has more to do with your mechanical design rather
than the electric circuit or software you are using.

Any M6/1mm pitch screw with a direct coupled  1.8º /200steps per rev  NEMA
to it has a theoretical mechanical precision of  0.005 mm

It all depends on what you are building, the forces you need to move and
the final result you are trying to achieve

rickysupreme, 2018-04-03:
One line hack? What is the hack? And what file do I open?

langwadt, 2018-04-03:
@rickysupreme add a line like this between like 398 and 399 in stepper.c to get steps output on pin 2 and 3 when moving X 

`st.step_bits = ((st.dir_outbits & (1<<X_DIRECTION_BIT)) ? (st.step_outbits<<1):st.step_outbits)&STEP_MASK;  `

rickysupreme, 2018-04-03:
I will try it when I get home in like 11 hours.  which pins will these output on?

langwadt, 2018-04-03:
step_x and step_y so pin 2 and pin 3

rickysupreme, 2018-04-03:
Ok, my Gcode only outputs one axis of movement. So this will still work?

langwadt, 2018-04-03:
yes as long as you only use the X axis it should work

rickysupreme, 2018-04-04:
Arduino: 1.6.8 (Windows 8.1), Board: "Arduino/Genuino Uno"

In file included from C:\Users\Supreme\Documents\Arduino\libraries\grbl/grbl.h:43:0,

                 from C:\Users\Supreme\Downloads\grbl-master - Number 2\grbl-master\grbl\examples\grblUpload\grblUpload.ino:27:

C:\Users\Supreme\Documents\Arduino\libraries\grbl/nuts_bolts.h:25:0: warning: "false" redefined [enabled by default]

 #define false 0

 ^

In file included from C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:24:0,

                 from sketch\grblUpload.ino.cpp:1:

c:\program files (x86)\arduino\hardware\tools\avr\lib\gcc\avr\4.8.1\include\stdbool.h:42:0: note: this is the location of the previous definition

 #define false false

 ^

In file included from C:\Users\Supreme\Documents\Arduino\libraries\grbl/grbl.h:43:0,

                 from C:\Users\Supreme\Downloads\grbl-master - Number 2\grbl-master\grbl\examples\grblUpload\grblUpload.ino:27:

C:\Users\Supreme\Documents\Arduino\libraries\grbl/nuts_bolts.h:26:0: warning: "true" redefined [enabled by default]

 #define true 1

 ^

In file included from C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:24:0,

                 from sketch\grblUpload.ino.cpp:1:

c:\program files (x86)\arduino\hardware\tools\avr\lib\gcc\avr\4.8.1\include\stdbool.h:43:0: note: this is the location of the previous definition

 #define true true

 ^

In file included from C:\Users\Supreme\Documents\Arduino\libraries\grbl/grbl.h:43:0,

                 from C:\Users\Supreme\Downloads\grbl-master - Number 2\grbl-master\grbl\examples\grblUpload\grblUpload.ino:27:

C:\Users\Supreme\Documents\Arduino\libraries\grbl/nuts_bolts.h:51:0: warning: "max" redefined [enabled by default]

 #define max(a,b) (((a) > (b)) ? (a) : (b))

 ^

In file included from sketch\grblUpload.ino.cpp:1:0:

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:85:0: note: this is the location of the previous definition

 #define max(a,b) ((a)>(b)?(a):(b))

 ^

In file included from C:\Users\Supreme\Documents\Arduino\libraries\grbl/grbl.h:43:0,

                 from C:\Users\Supreme\Downloads\grbl-master - Number 2\grbl-master\grbl\examples\grblUpload\grblUpload.ino:27:

C:\Users\Supreme\Documents\Arduino\libraries\grbl/nuts_bolts.h:52:0: warning: "min" redefined [enabled by default]

 #define min(a,b) (((a) < (b)) ? (a) : (b))

 ^

In file included from sketch\grblUpload.ino.cpp:1:0:

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:84:0: note: this is the location of the previous definition

 #define min(a,b) ((a)<(b)?(a):(b))

 ^

In file included from C:\Users\Supreme\Documents\Arduino\libraries\grbl/grbl.h:43:0,

                 from C:\Users\Supreme\Downloads\grbl-master - Number 2\grbl-master\grbl\examples\grblUpload\grblUpload.ino:27:

C:\Users\Supreme\Documents\Arduino\libraries\grbl/nuts_bolts.h:55:0: warning: "bit" redefined [enabled by default]

 #define bit(n) (1 << n) 

 ^

In file included from sketch\grblUpload.ino.cpp:1:0:

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:115:0: note: this is the location of the previous definition

 #define bit(b) (1UL << (b))

 ^

C:\Users\Supreme\Documents\Arduino\libraries\grbl\stepper.c: In function '__vector_11':

C:\Users\Supreme\Documents\Arduino\libraries\grbl\stepper.c:398:5: error: 'stepper_t' has no member named 'step_bits'

   st.step_bits = ((st.dir_outbits & (1<<X_DIRECTION_BIT)) ? 

     ^

exit status 1
Error compiling for board Arduino/Genuino Uno.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

rickysupreme, 2018-04-04:
Sorry I get errors compiling this it won't load onto the arduino.

rickysupreme, 2018-04-04:
In file included from C:\Users\Supreme\Documents\Arduino\libraries\grbl/grbl.h:43:0,

                 from C:\Users\Supreme\Downloads\GRBL-28byj-48-Servo-master\GRBL-28byj-48-Servo-master\grbl\examples\grblUpload\grblUpload.ino:27:

C:\Users\Supreme\Documents\Arduino\libraries\grbl/nuts_bolts.h:25:0: warning: "false" redefined [enabled by default]

 #define false 0

 ^

In file included from C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:24:0,

                 from sketch\grblUpload.ino.cpp:1:

c:\program files (x86)\arduino\hardware\tools\avr\lib\gcc\avr\4.8.1\include\stdbool.h:42:0: note: this is the location of the previous definition

 #define false false

 ^

In file included from C:\Users\Supreme\Documents\Arduino\libraries\grbl/grbl.h:43:0,

                 from C:\Users\Supreme\Downloads\GRBL-28byj-48-Servo-master\GRBL-28byj-48-Servo-master\grbl\examples\grblUpload\grblUpload.ino:27:

C:\Users\Supreme\Documents\Arduino\libraries\grbl/nuts_bolts.h:26:0: warning: "true" redefined [enabled by default]

 #define true 1

 ^

In file included from C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:24:0,

                 from sketch\grblUpload.ino.cpp:1:

c:\program files (x86)\arduino\hardware\tools\avr\lib\gcc\avr\4.8.1\include\stdbool.h:43:0: note: this is the location of the previous definition

 #define true true

 ^

In file included from C:\Users\Supreme\Documents\Arduino\libraries\grbl/grbl.h:43:0,

                 from C:\Users\Supreme\Downloads\GRBL-28byj-48-Servo-master\GRBL-28byj-48-Servo-master\grbl\examples\grblUpload\grblUpload.ino:27:

C:\Users\Supreme\Documents\Arduino\libraries\grbl/nuts_bolts.h:51:0: warning: "max" redefined [enabled by default]

 #define max(a,b) (((a) > (b)) ? (a) : (b))

 ^

In file included from sketch\grblUpload.ino.cpp:1:0:

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:85:0: note: this is the location of the previous definition

 #define max(a,b) ((a)>(b)?(a):(b))

 ^

In file included from C:\Users\Supreme\Documents\Arduino\libraries\grbl/grbl.h:43:0,

                 from C:\Users\Supreme\Downloads\GRBL-28byj-48-Servo-master\GRBL-28byj-48-Servo-master\grbl\examples\grblUpload\grblUpload.ino:27:

C:\Users\Supreme\Documents\Arduino\libraries\grbl/nuts_bolts.h:52:0: warning: "min" redefined [enabled by default]

 #define min(a,b) (((a) < (b)) ? (a) : (b))

 ^

In file included from sketch\grblUpload.ino.cpp:1:0:

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:84:0: note: this is the location of the previous definition

 #define min(a,b) ((a)<(b)?(a):(b))

 ^

In file included from C:\Users\Supreme\Documents\Arduino\libraries\grbl/grbl.h:43:0,

                 from C:\Users\Supreme\Downloads\GRBL-28byj-48-Servo-master\GRBL-28byj-48-Servo-master\grbl\examples\grblUpload\grblUpload.ino:27:

C:\Users\Supreme\Documents\Arduino\libraries\grbl/nuts_bolts.h:55:0: warning: "bit" redefined [enabled by default]

 #define bit(n) (1 << n) 

 ^

In file included from sketch\grblUpload.ino.cpp:1:0:

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:115:0: note: this is the location of the previous definition

 #define bit(b) (1UL << (b))

 ^

C:\Users\Supreme\Documents\Arduino\libraries\grbl\stepper.c: In function '__vector_11':

C:\Users\Supreme\Documents\Arduino\libraries\grbl\stepper.c:399:5: error: 'stepper_t' has no member named 'step_bits'

   st.step_bits = ((st.dir_outbits & (1<<X_DIRECTION_BIT)) ? 

     ^

exit status 1
Error compiling for board Arduino/Genuino Uno.

rickysupreme, 2018-04-04:
Maybe you could test it in arduino.ino?

langwadt, 2018-04-04:
try this , I don't have arduino so I can't try it
`st.step_outbits = ((st.dir_outbits & (1<<X_DIRECTION_BIT)) ? (st.step_outbits<<1):st.step_outbits)&STEP_MASK;`

rickysupreme, 2018-04-04:
Can you show a screenshot with the code before and after so I can see if I'm inserting it correctly?

rickysupreme, 2018-04-05:
Brilliant finally got to testing it and it seems to works better than my half baked hardware circuit as I expected. But I will be testing it with a 0.001mm digital dial indicator once it comes in the mail and will be able to post results eventually.  Now I need to work out how to get this thing to do microstepping as I seem to be limited to 0.02mm layer steps which is pretty good except the axis was kinda expensive so Id like to push it further if I can. Thanks for your help langwadt.

X3msnake, 2018-04-05:
Cool

Do post back your results :)

as of the code what were you doing wrong then?

If your motor is 0.72º that means your step resolution for a full turn is
500step/rev

If you move a step and your axis moves 0.02mm than that must mean you have
a 10mm pitch thread in your axel, is that correct?

rickysupreme, 2018-04-05:
The code works now, the first code provided did not. you will have to ask Langwadt. You are correct about the 10mm pitch thread, but i cant change it as it is integrated, It is also belt driven but maybe the ratio is 1:1.

rickysupreme, 2018-04-05:
I read somewhere that belt driven actuators are actually better than direct coupled. It takes 50 steps to move 1mm.

langwadt, 2018-04-05:
`st.step_outbits = ((st.dir_outbits & (1<<Z_DIRECTION_BIT))?(st.step_outbits>>1):st.step_outbits)&STEP_MASK;`

should put it step on pin 3 and pin 4 when moving Z

rickysupreme, 2018-04-05:
yeah strange, it dosn't work with the z direction bit. Only pin 4 is working

rickysupreme, 2018-04-05:
Ok it's working! It didnt work over two lines but maybe it was just a bad contact. Logic prevails!

rickysupreme, 2018-04-25:
Just got the digital micrometer indicator a few days ago. It seems to be able to hit the exact point to the 0.001mm 90 percent of the time after a full up/down motion. Positioning accuracy is not as precise (need to test more) but Repeatability is very good. Still want to move onto micro stepping next and need to check all my power supplies.

rickysupreme, 2018-04-27:
So it turns out I just need two pins that can output on command a constant high or low signal without any pulsing with this I will be able to halve the step angle and cut off the windings when needed according to the documentation provided.

X3msnake, 2018-05-04:
@rickysupreme 
nano DLP?
You'r building a UV printer? how did it turned out?

rickysupreme, 2018-05-04:
Almost done just need to iron out some kinks. 
Here is some data I recorded using half steps for 0.1 stepping during a dry test print cycle:

0.004
0.018
0.025
0.036
0.042
0.054
0.063
0.079
0.083
0.096
0.104
0.115
0.124
0.136
0.147
0.157
0.165
0.179
0.188
0.198
0.206
0.221
0.226
0.237
0.246
0.259
0.267
0.279
0.289
0.3
0.307
0.319
0.326
0.337
0.345
0.358
0.365
0.378
0.386
0.398
0.404
0.416

rheine4ever, 2020-08-15:
Hello, I just used a second arduino sitting in between GRBL and the VEXTA 5 Phase driver. Could not be simpler. I am not really an electronic man, but it seems to work in my setup. There could very well be timing/speed issues. To be honest I was surprised it works...
Here's the code:

int IN_GRBL_DIR = 3;
int IN_GRBL_STEP = 4;
int OUT_VEXTA_CW = 5;
int OUT_VEXTA_CCW = 6;

void setup() {
  pinMode(IN_GRBL_DIR,INPUT);
  pinMode(IN_GRBL_STEP,INPUT);
  pinMode(OUT_VEXTA_CW,OUTPUT);
  pinMode(OUT_VEXTA_CCW,OUTPUT);
}

void loop() {
  int step;
  int cw;
  int ccw;
  
  step = digitalRead(IN_GRBL_STEP);
  if (digitalRead(IN_GRBL_DIR)==LOW) {
    cw = step;
    ccw = 0;
  } else {
    cw = 0;
    ccw = step;   
  }
  digitalWrite(OUT_VEXTA_CW,cw);
  digitalWrite(OUT_VEXTA_CCW,ccw);
}

jafarsofian, 2020-09-21:
> Hello, I just used a second arduino sitting in between GRBL and the VEXTA 5 Phase driver. Could not be simpler. I am not really an electronic man, but it seems to work in my setup. There could very [...]

can you help me to add a  connection diagram for this method if you can

rheine4ever, 2020-09-21:
Yes, here it is
[image]

jafarsofian, 2020-09-21:
i have this   can i run stepper motor for this or need something??? i dont know some pins how itt work if you know please help me thanks for all things 
 
[image]

jafarsofian, 2020-09-21:
can you film a video for this connection ??? my best regards

rheine4ever, 2020-09-21:
It was a proof-of-concept setup. Here's a photo (a video would only add the motor actually turning its shaft :-))
[image]

jafarsofian, 2020-09-22:
ok i understand you thanks for all 💯 🥇

jafarsofian, 2020-09-22:
> It was a proof-of-concept setup. Here's a photo (a video would only add the motor actually turning its shaft :-)). The left arduino is connected, through USB, to my laptop running UniversalGcodeSender.jar which allows you to setup GRBL and send commands to it, making the motor turn. What your electronics board does I don't know. This is my proof of concept setup. Please ignore the background items. You should recognize the items in the diagram above.
> [image]

i do all connection very well but stepper not working so whats the problem i 
[image]

- [ ] attach picture to my connection

jafarsofian, 2020-09-26:
can you give me the first Arduino codes as gr-bl
?????????

rheine4ever, 2020-09-27:
hello, impossible to see of course where something is wrong. I think I would start with a 2phase motor and driver with the straight forward GRBL connections DIR/STEP and sending GCodes with UniversalGCodeSender. I that works ok, I would add the second arduino with the 5phase VEXTA and driver. Kind of divide & conquer. Hope this helps

eletronsystem, 2021-06-19:
> Para as gerações futuras, a solução para isso é
> 
> Inserir:
> 
> **st.step_outbits = ((st.dir_outbits & (1 << X_DIRECTION_BIT))? (st.step_outbits << 1): st.step_outbits) & STEP_MASK;**
> 
> entre as linhas 398 e 399 em stepper.c para obter a saída de etapas nos pinos 2 e 3 ao mover X

hello, i added this line between 398 and 399 to use my cw / ccw driver but i can't get it to work, would you be able to help?

m12xabood, 2023-09-20:
> can you give me the first Arduino codes as gr-bl ?????????

Hello, could you attach  the circuit diagram for the second picture ?

m12xabood, 2023-09-20:
> Yes, here it is (power to GRBL arduino out of scope :-)). Ofcourse you need one for each motor. Wire to GRBL accordingly, this is X-axis motor. Or...change converter code for multiple motor inputs! Again there will probably missed steps as this approach takes snapshots from the STEP pin and reproduces them on another pin CW/CCW leading probably to a somewhat different profile. But maybe not... So no guarantees. A hardware solution is probably appropriate, logically a SPDT relais should be able to do the job but would not be instantaneous with, for me, unknown consequences... Interesting subject!
> 
> [image]

Hello, could you attach the code file ?