grbl/grbl #1380: Pause and spindle stop

sfinexer, 2018-04-03:
Pressing a pause, the spindle doesn't stop. How to correct it?
grbl 0.9j

chamnit, 2018-04-03:
Feed hold only stops the feed, not the spindle. You'll need to enable parking mode in config.h or update to Grbl v1.1 and use the spindle override to stop the spindle.

closed 2018-04-03 by chamnit

sfinexer, 2018-04-03:
chamnit  not compiler grbl v1.1
`Arduino: 1.9.0-beta (Linux), Плата:"Arduino/Genuino Uno"

In file included from /home/sfinexer/Arduino/libraries/grbl/config.h:30:0,
                 from /home/sfinexer/Arduino/sketch_apr03a/sketch_apr03a.ino:1:
/home/sfinexer/Arduino/libraries/grbl/grbl.h:68:4: error: #error "Required HOMING_CYCLE_0 not defined."
   #error "Required HOMING_CYCLE_0 not defined."
    ^
/home/sfinexer/Arduino/libraries/grbl/grbl.h:108:4: error: #error "WCO refresh must be greater than one."
   #error "WCO refresh must be greater than one."
    ^
/home/sfinexer/Arduino/libraries/grbl/grbl.h:111:4: error: #error "Override refresh must be greater than zero."
   #error "Override refresh must be greater than zero."
    ^
exit status 1
Ошибка компиляции для платы Arduino/Genuino Uno.

`

chamnit, 2018-04-03:
Follow the compiling instructions exactly. Particularly the part that says to remove any previous installations of Grbl.

sfinexer, 2018-04-03:
grbl 0.9 is compiled without mistakes (previously having removed all old libraries). And v 1.1 isn't compiled, giving the error messages rendered above.