grbl/grbl #1381: Closed loop spindle rpm control

pffabrications, 2018-04-04:
Hi Iv been using GRBL with protoneer 3.51 shield and drv8225 stepper drivers on a chinese 3040 machine and its great however I want to be able to command the spindle via GCODE to run closed loop thus maintaining set RPM.

I have been developing a system to run on an arduino pro micro alongside GRBL but I am very much a Arduino coding novice
[https://github.com/pffabrications/Closed-Loop-SPINDLE-CONTROL-GRBL](url)

This is a cry for help!

I have used interrupts to detect spindle rpm via IR sensor and detect 0-5V PWM duty cycle from GRBL
The system does work to some extent but has many bugs. 

GRBL RPM Detection - 

Once PWM from grbl is detected the spindle starts and rpm can be controlled by GCODE - GREAT :)
However when spindle is turned OFF via GCODE the arduino does not detect 0 rpm and rpm set point stays at the last positive value received. 

Spindle RPM Detection-

Doesn't work as well as it could do. Possibly need to adjust samples taken and add smoothing/averaging to values. 

PID Loop - 

The way the PID loop reacts is not smooth so PID values could do with tuning but possibly a result of the less than ideal spindle rpm detection.

I would also like to add a switch to select manual/automatic mode, a potentiometer to set rpm in manual and fine adjust rpm in auto mode,  i2c 16x2 lcd display, overload protection and emergency stop provisions

If anyone can help me develop this it would be greatly appreciated.

langwadt, 2018-04-04:
use the spindle enable pin or add a timeout to the pwm input to detect off

pffabrications, 2018-04-05:
Thanks yes a timer would work thanks I'll try that. I think that is one of the issues with my code in that it is not running fast enough

mschorer, 2018-04-05:
Hey,

Maybe you want to have a look at
https://github.com/mschorer/grbl_mc_leonardo

I guess it does all/most things on your whishlist ...

mrX

<https://github.com/grbl/grbl/issues/1381#issuecomment-378853751>, or mute
the thread