grbl/grbl #1382: Connecting Laser

PCPayne, 2018-04-05:
Hi Everyone could anybody help me in telling how i can connect a laser to my arduino please? Ive ordered a driver for it. As was going to do it via a relay but decided against that idea. 

Im sure ive read somewhere that i can connect to the the spindle enable pin but unsure on how to turn it on off etc.

Thank You

PCPayne, 2018-04-10:
Hi yes im using a arduino uno and tb6560 drivers. I have bought a laser driver which has two cores for the power then two cores to control it.

PCPayne, 2018-04-10:
Thanks norbusz, i beleive i do yes. Whats the best way to do it?

norbusz, 2018-04-10:
Your welcome. For image engraving the best solution wouldbe the variable powered laser via d11 and gnd pin connection.

PCPayne, 2018-04-10:
Ill probably be doing mainly test but the odd image. Ill connect it up and give it a go. How would i go about testing it to see if it works?

norbusz, 2018-04-10:
Grbl firmware is already upload to your arduino boad? If yes you need to connect the laser as I mentioned on upside, after you need gcode sender and send the m3 s200 command for example, laser need to turned on. I propose to use LaserGRBL software.

PCPayne, 2018-04-19:
Hi nurbusz I still haven't had chance to try yet but I will soon.