grbl/grbl #1383: Arduino uno grbl variable spindle with S command doesn't work.

norbusz, 2018-04-06:
First of all sorry for my bad English, English is my 2nd language.

I have a problem wiht my arduino uno, the variable spindle method with 'S' command doesn't work. I have tried Grbl V1.1 & v0.9 & v0.8 versions.
I messured the voltage between D11 and GND pin with multimeter and the voltage everytime is just 0,27V, voltage keep in 0,27V when i write M3 SXXX command. 
Voltage is always 0,27V

I tried the same method on D12 pin too, and the result is same. 

(The variable spindle option is enabled in the CPU_map.h file,  #ifdef VARIABLE_SPINDLE)

Please help... I have not idea. 

Thanks in advance for all the help.

norbusz, 2018-04-06:
I'm using cnc shield v3.00 and arduino uno board, but when i measure the voltages i was not connected the shield to Uno board. 
Currently i can't attach exact pic, cuz i'm in work. 

I have these and, i use these for connection: 
And i have 2,5W laser with TTL driver board, i want to use this laser as variable laser power, i want to engraving some picture. 
[image]

I have no idea why not work the PWM.

Thanks in advance.

X3msnake, 2018-04-06:
The china shield has the wrong wired pins

It is for 0.8 before the change in pins to use the PWM

try connecting the pins directly to your arduino and test the spindle

chamnit, 2018-04-06:
@norbusz : M4 changes D13 and has nothing to do with PWM output by default. Disconnect your shield. Directly measure the output of your D11, D13 pins. You'll find that they work fine and that @X3msnake is right. The problem is your shield is not wired for v0.9+.

closed 2018-04-06 by chamnit

norbusz, 2018-04-06:
I had been messured directly on arduino uno, and the result is just 0.24, 0,27 volt. What is the solution ?

chamnit, 2018-04-06:
@norbusz : Try a different Arduino and/or test your digital pins with a simple Arduino program.

X3msnake, 2018-04-06:
@norbusz

You should measure without any shield connected to it are you doing that?

If so you should try burning a simple PWM sketch to your arduino, maybe you
have a faulty pin and try lighting a led with it to make sure it is not
faulty https://www.arduino.cc/en/Tutorial/Fading

norbusz, 2018-04-06:
Thanks for your proposal, i tried that. 
I change the D pin in the sketch to D11 that could try this pin works or not, and happend interesting things. 
The led cannot fading in D11 pin doesn't works, however when i connected to the D9 pin Led is fading and also fading in D13 pin. 

I'm now very confused. 
[image]

chamnit, 2018-04-06:
You need a new Arduino. Something fried your D11 pin. Likely your shield due it being wired incorrectly. Check the wiring paths for D11 and see where it goes to on the shield. Make sure it's not wired to ground or anything.

norbusz, 2018-04-06:
Yes you are right, i'm now certain that have a problem with my D11 and D12 pins on our arduino board. 
But i haven't idea what would be the reason. I try to buy new arduino uno board, i hope this will work correctly. I'm now sad a bit. 
Thank you for your kindly help and support in this case.

nobodyatall-irc, 2021-11-26:
#ifdef VARIABLE_SPINDLE it's not mean it is declared!
Declare it first!

#declare VARIABLE_SPINDLE 1

nobodyatall-irc, 2021-11-26:
I was in same problem, and nowhere could find help - so I leave help for next guys.

mdsultan11, 2022-06-21:
Hi Guys, i have a problem with **Missing X real time status report per 10 seconds. Interval XXX. Correct baud rate? Try Marlin response**. why is this so?
And also i have commented the Variable spindle to make my all 3 limit switches to work!!

Is there anything wrong i have done!! i need help