grbl/grbl #1387: About reading the grbl code

Rob2029, 2018-04-16:
Hello everyone, i need to attach the grbl code to an investigation paper I am doing for the university, I know that is a complicated code, but where can I locate the main file or what could I do to get the most important part of the code? Thanks!

whitetd, 2018-04-16:
Gcode it self ? Open with a txt editor. Code that process's gcode?, unless
its about coding not of much use.

Rob2029, 2018-04-16:
Not a gcode file, i need to showcase the grbl code in the paper, the .ino
file per se

datdadev, 2024-07-25:
@Rob2029 I am doing the thing that you were doing, may you share your paper for me to make a reference to see how you did it? Thank you!

Rob2029, 2024-07-25:
@datdadev I ended up copying links to the repository, you can use the system.c file to understand how GRBL works and maybe show a diagram based on it

datdadev, 2024-07-26:
I don't know if you have researched the algorithm of `stepper.c` and `planner.c` of the library? I find those were the main/important part to dive deeply into...