grbl/grbl #1390: 1 Axis 2 Motor

EragonPower, 2018-04-21:
Hi, I just started building a cnc, I bought a cnc shield and I connected all 4 motors, 2 for the X axis, 1 for the Y axis, and one for the Z axis, the problem is : how do I use a single axis for two engines?

thank you in advance whoever answers me

X3msnake, 2018-04-21:
Your shield must have a clone axis driver. It basically repeats the comands
for the axis you define normally with a jumper.

EragonPower, 2018-04-21:
where?
[image]

EragonPower, 2018-04-21:
where i need to put a jumper?
[image]

X3msnake, 2018-04-21:
https://www.youtube.com/watch?v=r1zi9vx3YP0

luisluna2, 2018-05-03:
Hello friends, one question: How can SWAP the ZAXIS motor to YAXIS motor?

I have Ramps 1.4 and Arduino MEGA

I try change pins in cpu_map.h but when I change the pins it does not work any changed that in this case is Y and Z, what I want to do is get 2 Y and 1 Z motors.

Thank you for your coments!

lalo-uy, 2018-05-03:
You can connect 2 motor controllers ( Pololu or what ever ) to the Y pins
to get this.
On some board there is an A motor controller that can be jumped to Y axis