grbl/grbl #1391: G53 Strange Behaviour, but only after running a file

alexcampbelledm, 2018-04-23:
Hello,
I am experiencing a frustrating and dangerous behaviour when issuing a G53 command after running a file. I am running UGS platform. I have home switches and what I am describing always happens with the machine homed and the MCS seeming to show normal values.
I like to use a parking spot after running a file to send my tool to the top of Z and then to the back of the table. I use the following;

G90 G53 G0 Z-10
G53 G0 Y600

This works fine when I initially set up my machine after homing, after jogging around the work area, but as soon as I run a file, the same code will crash my z axis down (positive value on my machine) and then try to send my machine to some high Y value that always crashes into the back of my machine. If I hit stop and perform a soft reset, it will work normally again. If I enable soft limits, it will always trigger an error that I am trying to send my machine outside of the envelope. 

I have tried sending 
G91 G53 G0 X0Y0Z0
to try to fool it into "snapping out of it" but that doesn't seem to work either. It will take that code fine and then when issued a G90 G53 G0 Y10, It will still try to send my machine off to some random place in space. I have looked at the verbose output, but it doesn't seem to hold any clues. It issues the same command before or after running a file, but this only happens after running a file. 

Any thoughts or experience? I would sure like to think I am missing something.

jahnj0584, 2018-04-23:
When you home the machine what do your work coords read as?

alexcampbelledm, 2018-04-23:
Hello, when zeroed, all coordinates read as 0,000