grbl/grbl #1395: Axis Z Switch Fail

berdamc, 2018-05-07:
Hi guys
I have a CNC router and i´m try to configure homing. The axis Y and X works fine, but i has problems in my end stops in axis Z.

All the end stops are NO, 
parameter $5 is 0
parameter $21 is 1
I used GRBL 0.9J and CNC shield V3.

Let´s go to the problem.

When i click $H in Universal GCode Sender, the axis Z go up, when the axis Z touch the end stops, the machine stop, until this point everybody ok, but after it my machine don´t work to make homing in axis Z.

if i don´t click in RESET SOFT, the axis Z break my end stops.

Anybody can help me ??

[image]

chamnit, 2018-05-07:
It's unclear what the problem is. You say that homing Z goes up and completes the homing cycle. Are you sure its not your hard limits that are triggering? They can be very finicky and trigger from very little electrical noise.

berdamc, 2018-05-07:
Tks Sonny.  I believe it's not end stops because grbl don't write 'hard
limit' on serial output.

berdamc, 2018-05-08:
Its possible end stops triggering and don't show 'hard limit' on serial output ?