grbl/grbl #1398: How to change PWM Frequency.

StuartB4, 2018-05-11:
Just wondered if anyone knows which part of the cpu_map.h file I need to edit to change the PWM
frequency from 1khz to 8khz. I found references to it but can't find anywhere where it actually tells you what to edit.
Just not sure what to do.

My thought was to comment out line 139 with // and remove the //'s from line 141.

138 // Prescaled, 8-bit Fast PWM mode.
139 #define SPINDLE_TCCRA_INIT_MASK ((1<<WGM20) | (1<<WGM21)) // Configures fast PWM mode.
140 // #define SPINDLE_TCCRB_INIT_MASK (1<<CS20) // Disable prescaler -> 62.5kHz
141 // #define SPINDLE_TCCRB_INIT_MASK (1<<CS21) // 1/8 prescaler -> 7.8kHz (Used in v0.9)
142 // #define SPINDLE_TCCRB_INIT_MASK ((1<<CS21) | (1<<CS20)) // 1/32 prescaler -> 1.96kHz
143 #define SPINDLE_TCCRB_INIT_MASK (1<<CS22) // 1/64 prescaler -> 0.98kHz (J-tech laser)

Any help would be greatly appreciated.

langwadt, 2018-05-11:
I'd expect: comment out line 143 with // and remove the //'s from line 141

StuartB4, 2018-05-11:
Thanks langwadt.

Is that all there is to it?
Edit those 2 lines , compile and reupload GRBL to Arduino.

chamnit, 2018-05-11:
Yes

closed 2018-05-11 by chamnit

StuartB4, 2018-05-11:
Ok, thankyou very much.