grbl/grbl #1405: $ commands not saving

KENSTV, 2018-05-30:
I'm getting various results on saving changes to the $100 and $101 commands, there does not seem to be any clear information on making changes actually happen, when I use the WINDOW in tools where the commands are listed I can change there and re start and the the commands for steps per mm show the changes 
but when I load the g code file the steps go back to previous settings, what is the proper way to change these settings in the bcnc tools menu on a protoneer raspberry hat board, only one time putting the change in the command line at the bottom of the window I got the acknowledgment of the change, another time changing the values in the window column and restart one value did change---what am I missing here and is there a clear guide out there for using bcnc-grbl ??

1Metalguru, 2018-05-30:
Try using the command line $100=val command, the Wiki says it saves it into eeprom immediately on execution, as long as you get the "OK" response.  It may be something flakey with your software which I am not familiar with.  I know setting it with GRBL Panel in Windows is flawless, so it's unlikely to be a bug in GRBL.

MG

KENSTV, 2018-05-30:
I believe it is possible to use the terminal program in bcnc since others have but due to lack of any documentation on using the terminal many have had problems making changes, some said only one command at a time can be changed and a restart is needed others have had it work, there is nothing on the bcnc or grbl github pages that clearly say how to use the terminal program in bcnc, I suspect it is a keyboard layout problem because when I select the command line and enter say $100=50 and hit enter
the command disappears from the command line, if I go to the panel above the command line I can select and edit the $100= and it goes to the next line, problem is it will not save it, my sd card access is disassembly of unit so using the grbl panel in windows will be my last resort --just got to be something simple like euro keyboard layout ?? although the enter key works on the raspberry pi,  the keyboard is a common one: Logitech Wireless Touch Keyboard K400 Plus with Built-In Touchpad --at a loss as to why it will not save the commands

chamnit, 2018-05-30:
@KENSTV : Grbl is a firmware and has no SD card. Just a serial interface. You are confusing what Grbl does and what bCNC does. Grbl works fine if you connect directly to it through a serial terminal. It will respond with an 'ok' after each $100=x command to tell you that it saved. That's it. If you are having issues with bCNC not being clear in what its doing, I would post on their GitHub page about your issue.

Also, the Grbl wiki pages clearly state that you have to stream settings to Grbl line by line. It's related to an EEPROM issue.

closed 2018-05-30 by chamnit