grbl/grbl #1407: Probe Not Working

micksulley, 2018-06-04:
This may well be my lack of knowledge, but I cannot get the probe to work
I am running grbl1.1 and bCNC 0.9.14, and trying to use the probe function.
I select the Probe tab and in the Probe box I leave the X and Y blank and enter 1 in Z, then click Probe.  The Z axis starts to move down as expected, but shorting the probe signal to earth has no effect.
Is there any way to read the probe signal from bCNC?  Have I missed something else?

Thanks
Mick

109JB, 2018-06-04:
> I select the Probe tab and in the Probe box I leave the X and Y blank and enter 1 in Z, then click Probe. The Z axis starts to move down as expected, but shorting the probe signal to earth has no effect.
Is there any way to read the probe signal from bCNC? 

Clearly not a Grbl issue but a bCNC issue.  You should ask this question to the bCNC developers or on the bCNC Github https://github.com/vlachoudis/bCNC

Additionally you mention that you are using Grbl 1.1, well, this is the Github for Grbl 0,9, so wrong place again,

As for Grbl 1.1, grbl reports probe pin status on normal status reports. If the probe pin is tripped the status report will include a Pn:P message indicating the probe pin is tripped.

micksulley, 2018-06-04:
Sorry to have posted in the wrong place but thanks for your help.  I have now solved it, embarrassed to say it was a wiring fault.

closed 2018-06-04 by micksulley