grbl/grbl #1408: problem Z axes

gianpiero80, 2018-06-11:
Hi guys,

i've got a woodpecker cnc 3018 with grbl 09.

only one things dont work, Z.

When i push upper arrow it run down, when i push down arrow it run down. it's ugual when i load a design that require to go up but go down.  

the machine dont have mechanical or motor problem, because ive try to change the connector z axis with anouther one and work good, up and down.

i hope you understand my poor english

109JB, 2018-06-11:
This could be a bad connection, a bad pin on the 328 mcu, or something else. We just don't have much to go on.

gianpiero80, 2018-06-12:
You are rigth man!

There was a mistake in the assembly! one  pin was out from the hole and therent electrical connections

thenk you