grbl/grbl #1409: In the Config.H file does this mean (if Enabled) GRBL will print the GCode line numbers on screen next to the GCode as it scrolls up the screen.

StuartB4, 2018-06-12:
// Allows GRBL to track and report gcode line numbers.  Enabling this means that the planning buffer
// goes from 16 to 15 to make room for the additional line number data in the plan_block_t struct
// #define USE_LINE_NUMBERS // Disabled by default. Uncomment to enable.

For instance, LaserGRBL streams the GCode up the screen but with no line numbers.
Would enableling this option produce line numbers next to the lines of GCode.
Or does would that need to be programmed in to LaserGRBL or whatever program is used to stream the GCode.

Thanks.

109JB, 2018-06-12:
the line numbers are in the G-code file. They are not "generated" by grbl or any GUI.  The CAM program you use could generate them when the G-code file is created

chamnit, 2018-06-12:
Or the streamer would generate the line numbers. It would also have to read and interpret them appropriately to be useful. I also don't recommend using this option for laser jobs, because it greatly decreases communication bandwidth, particularly during raster jobs.

closed 2018-06-12 by chamnit

StuartB4, 2018-06-12:
Ok, thats fine.
Thankyou very much.