grbl/grbl #1411: x-axis moves. y doesn't.

KonradHunter, 2018-06-23:
I'm new to Grbl. 
I'm using a v3 CNC shield and Arduino uno, and UGS to make a laser engraver.
When I send commands from UGS the y-axis motors don't react and the active status remains 'idle' It worked for a little while but stopped responding for no obvious reason. x-axis works fine. 
I've tried deleting grbl from the Arduino library and redownloading, reincluding, reuploading it. I've sent commands from Arduino Serial monitor instead of UGS. I've swapped motors and drivers. I purchased a new Shield. Results are the same.
Any Ideas?

Rob2029, 2018-06-24:
Try to download an older version of GRBl, and also re check every connection
There's a possibility that you're pushing the motor beyond the nominal forces, and it just doesn't have the force required  to move

KonradHunter, 2018-06-24:
Where do I download the archives? When I click on a link in archives it takes me to some lines of numbers and letters.

Rob2029, 2018-06-24:
don't know if they are available, give me your email, I'll check if I have
the files

chamnit, 2018-06-24:
There is no firmware reason why Grbl would stop making the y-axis stop. The CNC shields you have only have the z-limit pin and spindle enable switched. It shouldn't alter how it works. It's far more likely that the stepper drivers were incorrectly setup and either are not producing enough current or set way too high. Make sure your potentiometers on the drivers are set correctly.

KonradHunter, 2018-06-25:
Thanks I was able to get it worked out.  Still not sure what the issue was. I just redid everything.

Rob2029, 2018-06-25:
Glad you work it out, sometimes a mistake wiring or configuring can cause
headaches

closed 2018-06-25 by chamnit