grbl/grbl #1415: Grbl 0.9 ignoring limit switches during homing

dansperfect, 2018-07-15:
So the title says it all
 I have GRBL 0.9 on my desktop cnc. I have installed homing switch at the max of each axis. When i first set them up they worked fine. That was two days ago. The machine has sat there since because i meeded milling stock. So today i get the stock go to mill my piece but the homing is acting like limit switches are not there. They work as hard limits still but homing just keeps pushing into them. Any advice or help would be most appreciated.

Rob2029, 2018-07-15:
Re check the connections and also you could try switching them manually to
see how it behaves, try to run the homming via gcode, right now I don't
remember which command is used to home but it's and easy search

dansperfect, 2018-07-15:
I just did this and it works when I start the homing hit the limit switches they will stop. But when I let them go by them selves they are hitting the switches but just not stopping

dansperfect, 2018-07-15:
And now it just worked for some odd reason. Hopefully it stays working Thanks for the help Rob.

Rob2029, 2018-07-15:
That's odd, and if the switches are being actioned the software should
recognize it, maybe is some interference when the machine hits the switch,
try grbl 1.1

dansperfect, 2018-07-15:
How do I know if my board will support Grbl 1.1 it came shipped with 0.9?

Rob2029, 2018-07-15:
It should work, grbl 1.1 is the same,  I think the baudrate changes from
9600 to 115200, but in terms of configuration and functions it has the same
as the 0.9 (don't know if the 0.9 version has configuration for the
acceleration of every axis but the 1.1 does)

dansperfect, 2018-07-15:
Ok so trying to reprogram this thing I get a timeout when trying to upload the sketch. So I figured I would update the boot loader as there was a problem with the old one and now I get this. So I can't burn a new boot loader and It will not let me upload Grbl 1.1 

avrdude: programmer operation not supported

avrdude: Using SCK period of 10 usec
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

Rob2029, 2018-07-15:
Have you try it using the Hex files?

dansperfect, 2018-07-15:
no will try that now

dansperfect, 2018-07-15:
trying to use command line to upload hex file also gets me a timeout

Rob2029, 2018-07-15:
Sorry i can't help you with that

dansperfect, 2018-07-15:
Not a problem you helped me alot. I'm contacting Sainsmart now to see why I can't upload to the board and if there is maybe a jumper that needs to be in place or something

Rob2029, 2018-07-15:
Ok good luck \m/