grbl/grbl #1416: spindle/cutter touching copper causes hard limit trigger

brentdbell, 2018-07-18:
I am trying to debug a problem with a new system setup.  My machine seems to work fine unless I try to make a PCB - I am trying to router a PCB, but I found that the system was hard resetting shortly after starting to cut the copper.  I took a single sided copper clad board and put the copper side down - and the system works just fine cutting the FR4 side, but as soon as I try cutting copper I get hard limits triggering.  If I raise up the z-level so that the cutter is a few mm away from the copper - it goes through the full g-code without a limit being triggered.

The copper clad board is 2-way taped to a piece of FR4 that is 2-way taped to another piece of FR4 - so it is definitely not touching anything else...

I am using a Chinese spindle and VFD - I have both the ground wire (pin 4) from the spindle and the shield from the spindle cable connected to PE on my inverter.

All of my cables are shielded with the shield connected to ground/earth on the CNC power supply. I have both de-coupling caps and pull-up resistors on the limit switches - what's wrong?  Any help is appreciated!

Rob2029, 2018-07-19:
That's weird, if you can, check the voltage drop if there is, between the
spindle and ground when it is touching the copper, some odd parasite
inductance must be there, but I have no clue

brentdbell, 2018-07-20:
I ended up solving the hard limit problem by opening up the end of the spindle and attaching the ground wire which was missing.  I had already opened the connector and connected it there, but found when I opened the spindle it was not connected inside.  I actually bypassed the opto-couplers and tested it - now no false hard limit triggering at all.

I am now trying to figure out how to control the spindle speed...but that is really beyond the scope of this forum so will post elsewhere...this post was also...but hopefully will help someone with similar problems.

closed 2018-07-20 by brentdbell