grbl/grbl #1419: grbl 0.9 nothing on the screen

capnkirk10, 2018-07-31:
OK, total idiot here.  Just assembled a Woodpecker CNC carver and loaded the 0.9 code out to my laptop.  Followed all the instructions regarding installation.  When I execute the program, task manager sez the program is running, but there is nothing on the screen.  Uninstalled everything and started over.  Same results.  Any ideas y'all!

109JB, 2018-08-05:
Grbl does not have a "screen". It functions by receiving a serial stream of g-code and outputting appropriate stepper pulses. The "screen" part is the sending program which is separate from Grbl. Grbl itself doesn't care how the g-code is fed to it through the serail connection as long as it comes through properly and in the right format.

If you are having a problem getting that "screen" to display on your computer then it is an issue that you need to take up with the maker of that program. In any case it is not a Grbl issue.

capnkirk10, 2018-08-05:
There is a program in the software they send with the cnc mill.  It is called “GRBL Control”  When I executed it on my desktop system, there is a screen that comes up!  It has manual motion controls, the ability to draw, the ability to store, send, etc.  When I execute it on my laptop, which is to be used to drive the mill, nothing show up.  this is the program I am trying to get answers about, along with other things that don’t work right.  Several people have told me that GRBL does not have a screen.  The program that I am referring to obviously does, and I would like to get it to work properly.

Thanks for your help.  I am arguing with the maker of that program (I think) and getting nowhere fast.

Kirk Merley

From: 109JB 
Sent: Sunday, August 05, 2018 1:53 AM
To: grbl/grbl 
Cc: capnkirk10 ; Author 
Subject: Re: [grbl/grbl] grbl 0.9 nothing on the screen (#1419)

Grbl does not have a "screen". It functions by receiving a serial stream of g-code and outputting appropriate stepper pulses. The "screen" part is the sending program which is separate from Grbl. Grbl itself doesn't care how the g-code is fed to it through the serail connection as long as it comes through properly and in the right format.

If you are having a problem getting that "screen" to display on your computer then it is an issue that you need to take up with the maker of that program. In any case it is not a Grbl issue.

—

---
This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

papillion47, 2024-09-26:
Benim GRBL programımda da benzer sorun var. Program açıldığında "visualizer" yani görselleştirici penceresi de açılır ve yapılacak iş ve freze çakısının hareketleri oradan takip edilir, ilk ayarlar buna göre yapılırdı.  Bu şekilde aylarca güzel çalıştı. Sonra 1-2 ay ara verip yeniden çalıştırdığımda "visualizer" boş olarak karşıma çıktı. Neyin sebep olduğunu bir türlü keşfedemedim. Programı silip yeniden yüklemek de işe yaramadı. Yardımlarınızı bekliyorum. Teşekkürler..