grbl/grbl #1426: Power loss recovery

manisaranbala, 2018-08-20:
Grbl 1.1f power loss recovery. Define power loss pin. Possible?

109JB, 2018-08-20:
You are posting in the grbl 0.9 forum but state that you are using Grbl 1.1f.  
https://github.com/gnea/grbl/issues?q=is%3Aissue+is%3Aopen+sort%3Aupdated-desc

Also, you aren't very specific as to what you are looking for.  

In general, a power loss results in loss of all current position information which would require re-homing if you have homing switches installed, or if not then re-zeroing of your axes.

abdoale, 2018-08-21:
re-zeroing of your axes

manisaranbala, 2018-08-21:
My g-code total. Command 519520 so nunning 3 hour's  Running 30% job
running ac power cut. Than again run
Firs line running. How to solve my problem. (Cnc milling router for wood
carving) ac powe cut time last run line show my app how to jump last line
or skip gcode line

Norwegian-Gadgetman, 2018-08-21:
Even with re-homing you risk slight shifts in the carving because of the inherent flex in your machine. 
The only way to REALLY solve this issue is to avoid AC power loss altogether, and that means an UPS.