grbl/grbl #1429: Hacking GRBL to control a simple automaton

biasedlogic, 2018-08-25:
Hi,
this is NOT an issue with GRBL or a feature request, rather a question for those who know the ins and outs of the GRBL guts. @chamnit I hope you have a moment to comment ;-) 
I'm **hacking** GRBL for **non-CNC use.**

Short version (questions without background):
1. Is there any analog pin free that I can use? I believe to have tracked down A5/A6 as unused, but is there anything non-obvious that'd prevent me from setting up ADC for continuous conversion on one of these?
1. I need the feedrate adjusted on-the fly based on the converted ADC value, ignoring anything set by Fxxxx word. Can I just overwrite gc_block.values.f at the beginning of each gc_execute_line() or will it break something badly? Is there a better place for it? Changes in speed will be slow and it's ok if acceleration management fails for the next move.
1. can one re-enable movement commands in the init string? Especially homing. Actually, it's enough if homing in init string works. I know why it's disabled now.

**Background:**

Okay, the thing is, I have a simple automaton (1, maybe 2 steppers) that needs to repeat a very simple sequence of movements indefinitely. There's no danger to humans, no sharp blades turning, the whole thing can be stopped by putting a finger in the most critical place. This is by design so, the original drive was a highly geared DC motor and a nasty set of levers with enough torque to shear a crowbar, so the rest of the motion mechanics is on slip clutches etc.

I can control it using two Arduinos, one running GRBL and the other one feeding it with gcode. I want to get it done with one, so that I can use Woodpecker-style board as a standalone controller and have USB/serial free to reprogram the board in-system w/o issues.

I have hacked GRBL to continuously re-read the serial input buffer, which is now initialized with the required motion sequence, so the whole thing sorta works as intended. Incoming serial data is ignored except for override commands, so the thing is mostly dumb, but works in its basic aspect exactly as intended.

I'm left with two challenges:

Fine trimming the actual operating speed - it's not very critical, but has to be done by unskilled personnel and the proper operating speed changes depending on other factors, so I want a knob that anyone can turn -> hence analog in and internal computation of feed speed. 

I also would love to use a homing switch, now the first cycle is rough as the whole thing bumps into hard stops until the slip clutches adjust. This is of secondary importance though, as in normal operation the automaton can be thrown out of sync by audience interaction anyway, so the jerkiness of first cycle is aesthetics at best.

cri-s, 2018-08-25:
1.1 should have this analog speed setting. For 0.9 I have a diff but only
if AMASS is deactivated.

biasedlogic, 2018-08-25:
Okay, I've got it running. so what I did was:
1. Initialize the RX buffer with a string of commands that run the automaton. Each line gets an F-word with a dummy argument of "0", e.g. "F0G1X100\nF0G1X0\n".
1. The function to check how many characters are there in the buffer is modified to always return non-zero number.
1. The function checking buffer free space always returns zero.
1. The function to get data from buffer returns next character from the string and wraps to zero upon encountering zero-value character (the final character of a C string)
1. In main.c the ADC gets initialized and set to run continuously converting ADC5
1. in the parser, at the place where the F code gets interpreted, the value read from the buffer is ignored and the speed is computed based on the ADC value.

I didn't touch homing yet.

closed 2018-08-25 by biasedlogic