grbl/grbl #1430: GRBL XPRO V3 Doesnt stop when teaching homing, and doesnt stop on Z limit switch/.

adamfromayr, 2018-08-26:
Hey you guys!!!!!!!!!!!!!!!

Im stuck with my GRBL and x pro v3

My limit switches are all good and working

But when im trying to set up Homing for my CNC

The z Axis doesn’t stop when it hits the limit switch and ive enabled the Homing and Limit switches etc

It acts like its not hitting the z limit switch any ideas guys?

I made a video on [YouTube](https://www.youtube.com/watch?v=x1AyEQ3kUqI) hopefully get this sorted, I spent all of my Sunday trying before making video and posting!

jahnj0584, 2018-08-26:
Did you already try inverting the pins? Is it NC or NO circuits?

adamfromayr, 2018-08-27:
HI

Ive tried allsorts

Right now the z limit is on:: NC

And works 100$ with normal use

but no matter what i try when i press home on GRBL it never works, keeps juddering 

here is my layout?

MSG:'$H'|'$X' to unlock]
>>> $$
$0=67
$1=66
$2=3
$3=0
$4=1
$5=0
$6=1
$10=64
$11=410.336
$12=0.000
$13=0
$20=0
$21=1
$22=1
$23=4
$24=0.000
$25=96.374
$26=64
$27=195.794
$30=0
$31=0
$32=1
$100=900.000
$101=500.000
$102=700.000
$110=250.000
$111=500.000
$112=500.000
$120=5000.000
$121=33.000
$122=33.000
$130=250.000
$131=-398.336
$132=200.000
ok

adamfromayr, 2018-08-27:
**!!!!!!!!!!UPDATE!!!!!!!!!!!!!!!!!**

I changed the LIMIT SWITCH And now regonises it!!!!!!!!!!

But when it now HITS THE Z + Limit switch it the goes down and doesnt stop !!!!

**I do not have a Z - Limit switch, will i have to install a Z - Limit switch????**

jahnj0584, 2018-08-27:
I believe this is where the limit switch pin invert mask is applied. Look
at the wiki for the one that reverses only your Z

MeJasonT, 2018-12-13:
Why are your $$ config settings so different
[https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration](url)

Curious why your settings $0 & $1, I have never changed these on 3 machines i've built
$11 is so wrong, your minimum arc radius is 410mm seriously.
$12=0 ?
$24 homing speed is 0 - so no movement at all ?
$30 your max spindle speed is zero.

Ok dood i give up, try using default settings and then alter settings for any errors you find from there.

Use grbl panel [(https://github.com/gerritv/Grbl-Panel/releases) ] or [https://zapmaker.org/projects/grbl-controller-3-0/grbl-controller-main-screen-options/](url)

a program where you can use simple command instructions like xyz moves.

Get each axis to move using using single step and check they are moving in the correct direction.
then enable the hardware homing command.  try homing your machine - get ready to panic and hit stop if it goes mental.  

It should be reasonably straight forward to get to this point.

Then make step move, i suggest no more than 10mm, then measure the amount your axis is moving and calculate the pulses per step you need for your machine.