grbl/grbl #1431: Can't probe if probe already active

Harvie, 2018-08-29:
```
    G38.2 - probe toward workpiece, stop on contact, signal error if failure
    G38.3 - probe toward workpiece, stop on contact
    G38.4 - probe away from workpiece, stop on loss of contact, signal error if failure
    G38.5 - probe away from workpiece, stop on loss of contact
```

If probe already touches the workpiece, i can't do G38.3
If probe does not touch the workpiece, i can't do G38.5

I am working on novel (much faster) way of probing in bCNC, but it requires several probes immediately one after another. Please can you make it so that if i do G38.3 and there's already contact, the command will be silently ignored. Also G38.5 should be ignored if there already no contact.

G38.2 and G38.4 can remain the same if somebody needs "strict mode".

Thanks for considering this!

chamnit, 2018-08-29:
@Harvie : Sorry no. The g-code standard (that Grbl uses) states it is an error if the probe is already in target contact state upon programming the command. This applies to all G38.x commands. 

Can you describe what you are doing and what needs to be sped up? Maybe there is another way of doing it.

chamnit, 2018-08-30:
Why couldn't you alternate between G38.2 and G38.4? The g-code standard states that there is an error when the target state of the probe is active at the start of the command. Or look at the status report and get the active probe state before commanding the next probe command?

Harvie, 2018-08-30:
@chamnit that was first thing that i've tried. but there's the catch! when you move XY with probe active, you can never know what the next state of probe will be. if there's hill it will remain the same. if there's valley it will change.

chamnit, 2018-08-30:
Not sure why it wouldn't work. If it's a hill and the probe is still triggered, then move the z-axis up with G38.4 until loss of contact. Then G38.2 back down to get an accurate measurement. If a valley and the probe is not in contact, just G38.2 down to get a measurement.