grbl/grbl #1433: Allow to use ADC on probe pin

Harvie, 2018-09-01:
A5 pin which is used for probing is analog input. I want to use analog sensors for probing. Can you please make this possible? I know it's possible to use the comparator circuit to detect the treshold of analog voltage and output digital signal. But arduino already can do this, so why bother?

I suggest to add more possible values to `$6=0 (probe pin invert, bool)` option.

Eg.:
```
$6=0 (non-inverted digital)
$6=1 (inverted digital)
$6=-1 (inverted digital)
$6=123 (non-inverted analog with threshold of 123 (on scale 0-255))
$6=-123 (inverted analog with threshold of 123 (on scale 0-255))
$6=255 (non-inverted digital)
$6=-255 (inverted digital)
```

Rob2029, 2018-09-01:
what are the benefits of doing this?

Harvie, 2018-09-01:
This would enable us to easily use touch probes made using analog sensors. Currently i want to use probes based on load cells, piezo pickups and analog joysticks or computer trackpoints...

Rob2029, 2018-09-01:
Oh ok, like the ones usted in 3d printers for Z height?

Harvie, 2018-09-01:
Not exactly. In 3D priting world, they use additional electronics to preprocess the analog signal and make it digital. Also they only need to probe in one direction against flat perpendicular surface. I want to probe in 3D against surfaces, which can have slope, which can be bit tricky to achieve without complex probe design.

Rob2029, 2018-09-01:
Mmm i get ir, nice idea btw

MARIOBASZ, 2020-11-06:
any progress on this? have you used A5 as an analog input? There is a downside to setting the probe as analog (speed of response, maybe?)