grbl/grbl #1436: Probing precision bug

Harvie, 2018-09-21:
I've been doing some tests and i've found that faster i am probing, the later grbl stops the machine. This seems to be normal, since common sense is that faster you go, less precise will the measuring be due to inertia and parasitic capacity of probe cable and pin.

However i have strong belief, that there's also software bug in GRBL, which causes this! I will explain:

Let's say i am probing downwards on Z axis at 100mm/min and i get probe result -10.000mm, but when i probe on 400mm/min, i get -10.472mm.  And both of these measurements are very repeatable! I can switch back and forth between 100mm/min and 400mm/min and always get the same numbers for given feed speed!!! If i try gradualy increasing the speed from 100mm/min to 400mm/min, i can also see the probing error to increase accordingly from 0 to 0.472. If i go below 100mm/min the precision improves further, but not so drasticaly.

I think i can take the inertia out of play, because of repeatability of problem and the fact that my Z axis is on leadscrew preloaded by gravity, which pretty much prevents it from continuing the motion once leadscrew stops.

At first i was thinking that this is problem with arduino input pin being charged too slowly by internal pullup (20k ohm), so i've installed 500 ohm external pullup resistor to probe pin (= 40x more pullup current), but i still get the same -10.472mm reading no matter if i disconnect the additional pullup or not. So i can safely say that signal gets to arduino in time!

I think that **main problem is that GRBL does not check state of probe pin between each motor steps**, but rather several times a second. Which means, that faster you go the more steps are done between probe pin reads! This approach cannot lead to precise readings!

I don't know much about inner workings of GRBL. I am no uC guru and i haven't seen the code. Maybe this can be solved by using interrupt to handle probe pin. Maybe there are no interrupts left to do this, because they are used in some other parts of code. Maybe probe is handled by interrupt, but the interrupt is blocked by processing another interrupt related to motor stepping. I don't know.

Only thing i know is that probing is broken and inprecise in GRBL.

jahnj0584, 2018-09-21:
That's why in your macro you probe, raise up 1mm then probe again much
slower.

chamnit, 2018-09-21:
Grbl checks the probe state per step. It’s about as accurate as you can make it. What is more likely happening is inertial effects in your probe switch.

langwadt, 2018-09-21:
during probing GRBL checks the probe pin every step and copies the current position to the probe result when the probe pin is triggered, it might take a few step to cancel the motion (I'm not sure, I haven't checked how it is done) but the probe result is as exact as it can be

Harvie, 2018-09-21:
Just found this #1095 It's probably at least partialy caused by deceleration...
@langwadt so you say that probe result is precise, but position where machine is left after probing is not? i expected these to be the same (also i guess it's standardized by g-code to assume that).

chamnit, 2018-09-22:
The probe position is recorded when it is first detected as triggered per step interrupt. Grbl will then decelerate to a stop but will report the triggered position, not where it stopped at.

langwadt, 2018-09-22:
probing logs the trigger position and then stops within machine acceleration limits, the linuxcnc documentation say the same thing

chamnit, 2018-09-23:
Not really. An automated return to the probe point is omitted so it leaves more room for customization. You can easily insert a return command via g1 command, if you wanted it. If it were automated, you would not have the option of removing the return.

Harvie, 2018-09-24:
Nevermind, i've added option to automaticaly retract after probing directly in the bCNC. Thanks for explanation.

closed 2018-09-24 by Harvie

truglodite, 2021-02-27:
I also have this sorted with bCNC macros. However I imagine a large portion of use cases assume stop on contact does what it says, versus decel rather quickly after contact and stay there. I also think there are a lot of users who don’t use a sender with macro capabilities. From what I have seen, folks are ‘calibrating’ their probes to compensate for decel. At least most use a fixed feed and accel value for every probing. I also struggle to imagine what customization would make use of the stop position, or distance or whatever.

Harvie, 2021-02-27:
Maybe there could be additional g-codes for probing introduced to grbl. eg.:

G38.2 = traditional probing
G38.12 = probing with retraction