grbl/grbl #1438: removing coolant code to save space

beamzer, 2018-09-23:
I am trying to use Grbl 1.1f in CoreXY mode for a Laser plotter on an EleksDraw.
After switching to the Uno bootloader for the arduino nano on the EleksDraw,
the grbl code only just fits:
`Sketch uses 32114 bytes (99%) of program storage space. Maximum is 32256 bytes.
Global variables use 1633 bytes (79%) of dynamic memory, leaving 415 bytes for local variables. Maximum is 2048 bytes.`

To get a little bit more room, i would like to remove the coolant code, since that's not used
(i use only a 3.5W laser ;).
Is there an easy way to do that? The hard way would be to remove coolant_control.c and .h
and go through the code to remove all references until it compiles without error.

thanks in advance,
Ewald....

chamnit, 2018-09-23:
Try the current version, rather than the last release. I was able to free up a hundred bytes or so by refactoring a few things.

beamzer, 2018-09-28:
If i have to refactor, i might as wel do it in the latest release ;)

I removed most of the coolant stuff with a sed trick:
`mv coolant_control.c{,_park}`
`mv coolant_control.h{,_park}`
`sed -i .bak '/[cC][oO][oO][lL][aA][nN][tT]/s/^/\/\/E /g' *.h *.c`

and then went in to fix the broken "if" blocks. That gave me:
`Sketch uses 30128 bytes (93%) of program storage space. Maximum is 32256 bytes.`
`Global variables use 1630 bytes (79%) of dynamic memory, leaving 418 bytes for local variables. Maximum is 2048 bytes.`

Would be nice if in a future release it was possible to disable unneeded functions
like spindle speed, coolant and homing

closed 2018-09-28 by beamzer

franchouille77, 2022-05-12:
Hi, I had the same problem so I did it and removed the coolant and spindle code. Thanks for this solution It works just fine.
The code is here: [https://github.com/franchouille77/grbl-light-version](url)