grbl/grbl #1439: GRBL not responding after 1st command

arjtilley, 2018-09-23:
Hello, probably a typical noob issue here but here goes:

I have assembled a CNC 2417 kit and I am trying to get it moving. It came with a generic CNC controller board with a sort of Arduino-clone type module plugged into it with a Mega328P chip on it. It came pre-programmed with Grbl v0.8.

I can connect to the board using Arduino IDE's Serial Monitor, baud rate 115200 and I get the message Grbl v0.8 ['$' for help], so Grbl is flashed correctly on the board.

When I send commands however, it stops responding correctly after the first one. For example, I send G0 X1.0 and the X axis moves as expected. Then I send another G0 X1.0, I get a response something like (1/1) ok, but the machine does not move. Strangely, if I then send a G0 X-1.0, the machine moves back. But the subsequent command only works if the direction is reversed, the magnitude of the move does not matter.

I have tried using different line endings, no line ending, newline, carriage return (which grbl is supposed to use), and newline+carriage return, but I get the same thing. Also, sending a $X command to unlock grbl does not make a difference either. The only way is to hard reset the board after each command. I have also tried sending jogging commands using Candle, but these do not seem to work at all. Strangely, however, the spindle controls, Z axis probe and safe position commands do work correctly in Candle.

I should add that I have tested all of the servos and they all function, but they all have the same problem as above. They also all seem to constantly make buzzing noises of varying pitches when idle, don't know if this is normal, I also have a 3d printer that uses the same type of servos and it doesn't make these noises when idle.

Any ideas?

109JB, 2018-09-24:
Read about absolute and incremental mode in G-code.  They are G90 and G91 respectively.

Rob2029, 2018-09-24:
I have had  that problem when for example I use Chillipeppr and the switch
to Universal Gcode Sender, my arduino has grblv1.1 but that problem appears
on UGCS exactly as you described, I imagine that is an issue with the
handshake between the PC and grbl, try to reset everything.
With the noises of the motors, it could be various things, bad drivers, bad
current limitation, bad calibration of the speed and acceleration.. Try
checking all, and if you can try different drivers also do that

chamnit, 2018-09-24:
I agree with @109JB, this is not a comm issue. It's a misunderstanding of how g-code works. The default is moving in absolute coordinates. When the second command with the same target is sent, it'll be ignored because it's already there.

closed 2018-09-24 by chamnit