grbl/grbl #1440: Homing with 1 Axis Machine

ssalinasf, 2018-09-26:
Hi, I don't understand how change firmware to 1 axis machine, usually tutorials are for 2 axis machine.
```
// NOTE: Defaults are set for a traditional 3-axis CNC machine. Z-axis first to clear, followed by X & Y.
#define HOMING_CYCLE_0 (1<<Z_AXIS)                // REQUIRED: First move Z to clear workspace.
#define HOMING_CYCLE_1 ((1<<X_AXIS)|(1<<Y_AXIS))  // OPTIONAL: Then move X,Y at the same time.
// #define HOMING_CYCLE_2                         // OPTIONAL: Uncomment and add axes mask to enable

// NOTE: The following are two examples to setup homing for 2-axis machines.
// #define HOMING_CYCLE_0 ((1<<X_AXIS)|(1<<Y_AXIS))  // NOT COMPATIBLE WITH COREXY: Homes both X-Y in one cycle. 

// #define HOMING_CYCLE_0 (1<<X_AXIS)  // COREXY COMPATIBLE: First home X
// #define HOMING_CYCLE_1 (1<<Y_AXIS)  // COREXY COMPATIBLE: Then home Y
```
 That is the code to modify, to only home X axis, should I have the next code o should I do anything else?

```
// NOTE: Defaults are set for a traditional 3-axis CNC machine. Z-axis first to clear, followed by X & Y.
//#define HOMING_CYCLE_0 (1<<Z_AXIS)                // REQUIRED: First move Z to clear workspace.
//#define HOMING_CYCLE_1 ((1<<X_AXIS)|(1<<Y_AXIS))  // OPTIONAL: Then move X,Y at the same time.
// #define HOMING_CYCLE_2                         // OPTIONAL: Uncomment and add axes mask to enable

// NOTE: The following are two examples to setup homing for 2-axis machines.
// #define HOMING_CYCLE_0 ((1<<X_AXIS)|(1<<Y_AXIS))  // NOT COMPATIBLE WITH COREXY: Homes both X-Y in one cycle. 

#define HOMING_CYCLE_0 (1<<X_AXIS)  // COREXY COMPATIBLE: home X
// #define HOMING_CYCLE_1 (1<<Y_AXIS)  // COREXY COMPATIBLE: Then home Y
```

chamnit, 2018-09-26:
`#define HOMING_CYCLE_0 (1<<X_AXIS)`

Comment everything else.

closed 2018-09-26 by chamnit

awatar100, 2019-06-01:
Is this work ? .....
I build one axis machine - X. What i have to do to perform homing ? I change some code in grbl ver 0.9j But still i've got Homing fail....
I have change config.h to have only X axis in homing :
`// NOTE: Defaults are set for a traditional 3-axis CNC machine. Z-axis first to clear, followed by X & Y.
#define HOMING_CYCLE_0 (1<<X_AXIS)                // REQUIRED: First move Z to clear workspace. bylo (1<<Z_AXIS)   
//#define HOMING_CYCLE_1 ((1<<X_AXIS)|(1<<Y_AXIS))  // OPTIONAL: Then move X,Y at the same time.
// #define HOMING_CYCLE_2                         // OPTIONAL: Uncomment and add axes mask to enable`

.. I connect NC switch to X input on arduino nano ( when i click it i see change in grbl console) : 
`<Alarm,MPos:-2.000,0.000,-1.000,Lim:111>`  -> it change when activated 
`<Alarm,MPos:-2.000,0.000,-1.000,Lim:101>`  ->
My setup for machine
`$$
$0=10 (step pulse, usec)
$1=25 (step idle delay, msec)
$2=0 (step port invert mask:00000000)
$3=0 (dir port invert mask:00000000)
$4=0 (step enable invert, bool)
$5=1 (limit pins invert, bool)
$6=0 (probe pin invert, bool)
$10=17 (status report mask:00010001)
$11=0.010 (junction deviation, mm)
$12=0.002 (arc tolerance, mm)
$13=0 (report inches, bool)
$20=0 (soft limits, bool)
$21=1 (hard limits, bool)
$22=1 (homing cycle, bool)
$23=1 (homing dir invert mask:00000001)
$24=25.000 (homing feed, mm/min)
$25=25.000 (homing seek, mm/min)
$26=250 (homing debounce, msec)
$27=1.000 (homing pull-off, mm)
$100=6395.590 (x, step/mm)
$101=1.000 (y, step/mm)
$102=1.000 (z, step/mm)
$110=260.000 (x max rate, mm/min)
$111=260.000 (y max rate, mm/min)
$112=260.000 (z max rate, mm/min)
$120=20.000 (x accel, mm/sec^2)
$121=5.000 (y accel, mm/sec^2)
$122=5.000 (z accel, mm/sec^2)
$130=130.000 (x max travel, mm)
$131=10.000 (y max travel, mm)
$132=10.000 (z max travel, mm)
`
What else i have to do :/