grbl/grbl #1441: UGCS Works with one Arduino board, but not another

cdaileyjr, 2018-09-26:
With one Android Uno board the UGCS interacts appropriately.  However with another board UGCS  acts differently.

Once the USB to the Arduino  is connect to the computer and the "open" button is selected I There is a blank Active State Under Machine State.  This is the case even when I hit the jog buttons under Machine Control even though the commands are acknowledged under the Console and the Work Position X, Y and Z are adjusted accordingly.

The bottom line is that my CNC machine axis do not responding to the UGSC commands.

What could be wrong with the Arduino board to cause it not to communicate properly with UGSC?  

Thank you for any helpful suggestions.

Thank you,

Calvin

109JB, 2018-09-28:
Are both arduinos using a Grbl version that is compatible with the version of UGCS you are using?

cdaileyjr, 2018-11-02:
Yes, they are both GRBL compatible