grbl/grbl #1442: Cycle start to request stream

Harvie, 2018-10-05:
I think that hitting hardware "cycle start" button in idle mode should send some message over serial. So we can make g-code streaming software to start streaming once it get this message.

terjeio, 2018-10-06:
Parsing Ctl: part of real-time report (or Pn: in later versions) is not working? I have added parsing of Pn: to my GCode sender to good effect, however it requires the signal to stay on long enough to be catched by a report.

Harvie, 2018-10-06:
@terjeio I don't really like this idea. stream request should be sent even for short button press independently on status report interval.

terjeio, 2018-10-06:
@Harvie 

> How do i get the real time report?

Send a ? character. See the [wiki](https://github.com/grbl/grbl/wiki/Interfacing-with-Grbl) for details. Or did you mean how to add it to the report?
> Is that already in upstream grbl/grbl?

Yes.
> how can i enable it.

It is enabled by #define REPORT_FIELD_PIN_STATE from v1.1 (default), by #define REPORT_CONTROL_PIN_STATE in earlier versions (not default).
> I suppose it only shows the pin number and state. problem is that pin number of "cycle start" button might be different on various platforms. I would like to implement this in streamer with future ports in mind.

Pre v1.1: it is the control register contents as a number, a bitfield. from v1.1: the various bits encoded in a string with each character representing the function, eg. 'S' for Cycle Start. If you are going to support pre v1.1 versions you may need to add a configuration for how to parse the bitfield.
> Also. Can i be sure not to get the message twice?

No, state handling in your sender has to take care of that.  As long as a signal is active it will be reported.

Harvie, 2018-10-06:
> Send a ? character. See the wiki for details.

Yes, but that's not realtime

> Or did you mean how to add it to the report?

i have v1.1 with $10=3 and  it does not show Pn: or Ctl:

```
<Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|WCO:20.000,0.000,0.000>
<Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|Ov:100,100,100>
```

Harvie, 2018-10-06:
Aha. It showed only with that button active:
`<Idle|WPos:-20.000,0.000,0.000|FS:0,0|Pn:S>`

Harvie, 2018-10-06:
@terjeio
OK. Great. I've managed to modify bCNC to start streaming when START CYCLE pin is activated. I am happy with this!

Can we make the grbl to send the report immediately when the pin status is changed during idle?

Harvie, 2018-10-06:
This is the bCNC commit that adds this feature: 4a77780c9a01e2d63af036784097a404c510a336

terjeio, 2018-10-07:
@Harvie 
> Can we make the grbl to send the report immediately when the pin status is changed during idle?

Add
```
bit_true_atomic(sys_rt_exec_state, EXEC_STATUS_REPORT); 
```
to the control ISR.

Harvie, 2018-10-07:
Is it possible to make this grbl default?

Harvie, 2018-10-07:
> What's the problem with polling the status?

The fact that the press of the button might be shorter than polling interval and therefore go unnoticed. This will get us that annoying unresponsive feel of buttons in elevators made in last century.