grbl/grbl #1445: Not issue, but a great idea of speed control for CNC

EMinkoff, 2018-10-08:
Hi all. First I think this is a great project. I admire the guys who developed it. 
About the idea: 
What if we put a current meter for stepper motors and use it for feedback? Something like this:
[http://www.ti.com/lit/ds/symlink/ina3221.pdf](url)
This is a 3-channel voltage/current sensor, capable of measuring up to 26 volts and 3.2 amps with stock shunt resistors. More with smaller shunts. It has I2C bus and work great with Arduino.
I'm more in hardware and can't implement sensor's library in GRBL. I imagine something like this: 
"Do some steps for a second, measure current, if current is less than xxx (user definable max speed) then increase steps/s"
The lighter material for cutting, the lower current and the greater speed and vice versa.

[image]

chamnit, 2018-10-08:
The main problem is IO pins. The more sensing you have, the more pins you need. That said, modern motor controller ICs are now coming with SPI ports to set them up and provide data like current. There are now use cases for this in industry, like the new Prusa printers that use the current sense to detect limits, rather than using switches.

langwadt, 2018-10-08:
stepper drivers are constant current so measuring supply current doesn't give you that much useful information except maybe a stall at which point it is too late anyway.  afaik the new Prusa uses a new stepper driver with all kinds of trickery build in

EMinkoff, 2018-10-08:
INA3221 use only 2 pins connected to arduino for i2c and can be used together with other i2c connected stuff. Each driver's power goes trough the sensor, witch have continuous reading mode.