grbl/grbl #1453: HOMING FAIL

marcosyoggy, 2018-10-23:
Estou usando:  
-Python 2.7.15
-Pyserial 3.4
-BCNC
-UGS 0.9 e/ou 1.1
-Arduino Uno original
-Windows 7 32 bits

  A simulação usando apenas o Arduino conectado ao PC, eu consigo mover os eixos manualmente ,mas ao usar o comando $H ocorre o Alarm: homing Fail.

X3msnake, 2018-10-24:
Mas mexe-se a tentar fazer o homing?
Precisamos que respondas aqui com as tuas configurações de ligação dos homing switches, e fotos do equipamento assim como a lista e os settings de firmware que podes emitir no UGS escrevendo $$ na linha de comandos

marcosyoggy, 2018-10-24:
A primeira vez que aconteceu as configurações eram:
- fim de curso conectados às entradas +X,+Y,+Z, cabos com isolamento e capacitores de 'debouncing'
- Arduino Uno e Shield
- Fonte de 24V
- Programas usados: BCNC, UGS v1.1 
- Firmware:
$0=10
$1= 255
$10= 3
$20= 0
$21= 1
$22= 1
$24= 100
$25= 500
$26= 255
$27= 8
$100= 20(GT2)
$101= 20
$102= 100(fuso)
$110= 750
$111= 750
$112= 250
$120= 30
$121= 30
$122= 30
$130= 0
$131= 0
$132= 0

    Com tudo ligado, consigo mover os eixos individualmente sem problemas, mas ao comandar o $H,onde o eixo Z faria o 1º movimento, no motor do eixo Z é perceptível a energização, mas acontece apenas um tranco, mas sem movimento, e aparecem os  alarmes de 'homing fail'.
     Em seguida , retirei o Shield e testei com o mesmo resultado, reinstalei o GRBL mudando as versões e nada, a ultima coisa feita foi formatar o PC, reinstalar os programas e testar  apenas com o arduino conectado ao PC, simulando com o UGS, primeiro com a versão '0.9j' e depois a '1.1', com o mesmo resultado: consigo mover os eixos ,tendo os valores mostrados no painel de comando, mas quando realizo o comando '$H', independente da versão do UGS, o valor do eixo Z muda para um valor aleatório,como se estivesse movido, e aparece o alarme 'Homing Fail'.
     Independente, se na simulação ou com a CNC montada, o eixo Z não se move, percebe-se que o motor é energizado,mas não se move. Obrigado pela atenção e ajuda.

X3msnake, 2018-10-24:
> se faz o mesmo sem nada ligado é estranho.
> talvez necessites limpar a eeprom do arduino.
>
> tens algum outro arduino que possas usar como teste? se tiveres instala o
> 1.1 com as definições por omissão
>
> como fizeste o flash do arduino? a partir do IDE do arduino ou com o hex
> pré-compilado?
>
>
>
> No dia quarta-feira, 24 de outubro de 2018, marcosyoggy <

outra coisa tens as ligações dos endswitches configurados com NC ou NO?

coloca a lista completa de definições. faltam ali na tua lista informações
sobre os endswitches e as configurações de homing.

marcosyoggy, 2018-10-24:
- Com relação a EEPROM sempre limpo antes de instalar o GRBL;
- O GRBL eu instalo a  partir do zip ,com a IDE do arduino, não sei como usar o 'hex-precompilado';
- Os 'Fim de curso' são NO;
- Como eu uso  o Shield, faço a mudança no arquivo 'config.h' para inverter os pinos D11 e D12;