grbl/grbl #1454: Y axis one motor stops during homing

adamhaze8, 2018-10-24:
Ive been trying to get my new router to run grbl properly for a while now, and when I thought I finally had it, another extremely odd issue came up. It all worked for the first few homings,  but soon this started to happen. when I manually move the y axis any distance in grbl controller, everything works just fine, but when I want it to home the axis, the two nema 23 steppers run just fine for a while until one stops and it all jams before reaching the endstop. The one that stops is the original y axis port motor, not the cloned one on my cnc shield 3.0. I already checked wiring and that the stepper drivers were good. It is strange that this only occurs during homing. Im using an arduino uno with grbl 0.9. Im honeslty clueless as to what is happening. Any suggestions would be greatly appreciated.

X3msnake, 2018-10-24:
u sure there is no mechanical issue on that axel? can you reach the switch
by manual jogging?

how are you connecting to the endstops NC? NO?

what are your settings?

adamhaze8, 2018-10-24:
When jogging, the y axis moves around with no issues and can make contact with the endstop. It’s only during homing that it starts travelling properly for a second then the one y axis motor stops. The end stops are normally closed (nc) so I have the logic inversed($5=1). The homing worked fine the first few times then had issues as if something degraded.

X3msnake, 2018-10-24:
I would recheck all the electric connections and try running the mottors
uncoupled from the axis just to rule out mechanical problems.

Did you tune the drivers reference voltages??

à(s) 17:03:

adamhaze8, 2018-10-24:
I know the problem isn’t the drivers because I tested for that by switching the drivers on the board. It seems to be the port for the drivers on the shield or the wiring to the motor or the motor itself. But I checked the wiring it’s all good

adamhaze8, 2018-10-24:
Any chance there’s something wrong with the Arduino transferring all the electricity because I notice that the x and y home at the same time rather than when I’m jogging it manually

X3msnake, 2018-10-24:
do you have access to a logic analizer?

what motors are u driving? would a single driver have enough power to move
both motors for testing pourpouses?

109JB, 2018-10-25:
Microstepping is something the stepper motor driver does.  All grbl knows is how many pulses (steps) to send per mm.  

As for motor driver settings, it sounds like you either have DRV8825 or A4988. The directions for setting them up is at the following links

DRV8825 - https://www.pololu.com/product/2133
A4988 - https://www.pololu.com/product/1182